pauldemarco/flutter_blue #530: Unhandled Exception: MissingPluginException(No implementation found for method setLogLevel

MbarkErras, 2020-04-06:
E/flutter (30606): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: MissingPluginException(No implementation found for method setLogLevel on channel plugins.pauldemarco.com/flutter_blue/methods)
E/flutter (30606): #0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:154:7)
E/flutter (30606): <asynchronous suspension>
E/flutter (30606): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:329:12)
E/flutter (30606): #2      FlutterBlue.setLogLevel (package:flutter_blue/src/flutter_blue.dart:161:20)
E/flutter (30606): #3      new FlutterBlue._ (package:flutter_blue/src/flutter_blue.dart:27:5)
E/flutter (30606): #4      FlutterBlue._instance (package:flutter_blue/src/flutter_blue.dart:29:38)
E/flutter (30606): #5      FlutterBlue._instance (package:flutter_blue/src/flutter_blue.dart:29:22)
E/flutter (30606): #6      FlutterBlue.instance (package:flutter_blue/src/flutter_blue.dart:30:38)
E/flutter (30606): #7      bluetoothScanner (package:android_ble_code/bluetoothrecorder.dart:13:41)
E/flutter (30606): #8      _setupForegroundServiceCallbackChannel.<anonymous closure> (package:foreground_service/foreground_service.dart:208:50)
E/flutter (30606): #9      _setupForegroundServiceCallbackChannel.<anonymous closure> (package:foreground_service/foreground_service.dart:204:41)
E/flutter (30606): #10     MethodChannel._handleAsMethodCall (package:flutter/src/services/platform_channel.dart:409:55)
E/flutter (30606): #11     MethodChannel.setMethodCallHandler.<anonymous closure> (package:flutter/src/services/platform_channel.dart:377:54)
E/flutter (30606): #12     _DefaultBinaryMessenger.handlePlatformMessage (package:flutter/src/services/binding.dart:218:33)
E/flutter (30606): #13     _DefaultBinaryMessenger.setMessageHandler.<anonymous closure> (package:flutter/src/services/binding.dart:252:13)
E/flutter (30606): #14     _DefaultBinaryMessenger.setMessageHandler.<anonymous closure> (package:flutter/src/services/binding.dart:251:38)
E/flutter (30606): #15     ChannelBuffers.drain (dart:ui/channel_buffers.dart:188:21)
E/flutter (30606): #16     _DefaultBinaryMessenger.setMessageHandler (package:flutter/src/services/binding.dart:251:23)
E/flutter (30606): #17     MethodChannel.setMethodCallHandler (package:flutter/src/services/platform_channel.dart:375:21)
E/flutter (30606): #18     _setupForegroundServiceCallbackChannel (package:foreground_service/foreground_service.dart:204:20)
E/flutter (30606): #19     _runMainZoned.<anonymous closure>.<anonymous closure> (dart:ui/hooks.dart:241:25)
E/flutter (30606): #20     _rootRun (dart:async/zone.dart:1184:13)
E/flutter (30606): #21     _CustomZone.run (dart:async/zone.dart:1077:19)
E/flutter (30606): #22     _runZoned (dart:async/zone.dart:1619:10)
E/flutter (30606): #23     runZonedGuarded (dart:async/zone.dart:1608:12)
E/flutter (30606): #24     _runMainZoned.<anonymous closure> (dart:ui/hooks.dart:233:5)
E/flutter (30606): #25     _startIsolate.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:301:19)
E/flutter (30606): #26     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:168:12)
E/flutter (30606):

ramsestom, 2020-04-06:
Same issue when trying to run the example app on an android device with android 5.1 (No implementation found for method setLogLevel on channel plugins.pauldemarco.com/flutter_blue/methods)
This error seems to be related to android V2 plugin support (https://github.com/flutter/flutter/wiki/Upgrading-pre-1.12-Android-projects) as the debug console prints the following error :

> Tried to automatically register plugins with FlutterEngine (io.flutter.embedding.engine.FlutterEngine@3e117a2f) but could not find and invoke the GeneratedPluginRegistrant

However, testing the exact same example app on an android 8.0 device, this app is runing fine so it is a bit weird.

MbarkErras, 2020-04-06:
I am writing my own plugin :dancer:

ramsestom, 2020-04-06:
The fact you are writing your own plugin do not solve this issue for this plugin. 
So no, it isn't solved

pauldemarco, 2020-04-06:
A fix for this will be included in the next update.

ramsestom, 2020-04-06:
This fix has not been comited to the github repository yet? Because I am using the latest version of the master repository so if the fix is supposed to be in it, it is not working

pauldemarco, 2020-04-06:
Not yet.

MbarkErras, 2020-04-07:
@pauldemarco to add some insight the problem might be bacause I am running it inside flutter_foreground_service

adailydose, 2020-07-31:
@MbarkErras can you share the plugin you wrote?

sasp1, 2020-08-15:
Is there any update on this issue? I experience the issue when calling FlutterBlue.instance in a top level function.

Jbz797, 2021-01-28:
Some news about this issue ?

vastwebindia1, 2021-07-06:
> A fix for this will be included in the next update.

issue solved ?

DanielPiflaks, 2021-08-16:
I'm facing the same issue when I'm trying to run it with workmanager

aanantt, 2022-05-01:
**2 may 2022** and still the same issue

ihorkozar, 2022-06-03:
same issue

phisac12, 2022-07-12:
i'm having same problem, maybe this not is being updated :/

krzysztofkuczek, 2023-02-28:
Hi, same problem, any update?

rdmasaile, 2023-04-30:
is there a solution to this problem? I encountered this problem when trying to scan for bluetooth devices using flutterBlue package in flutter background service.

simranss, 2023-05-02:
I encountered this problem when I was simply calling startScan()

Exception: MissingPluginException(No implementation found for method setLogLevel on channel flutter_blue_plus/methods)

tylermcdonald, 2023-05-18:
> I encountered this problem when I was simply calling startScan()
> 
> Exception: MissingPluginException(No implementation found for method setLogLevel on channel flutter_blue_plus/methods)

I was running into this, but I found that stopping the app entirely and restarting it fixed the issue

stefan-sherwood, 2023-06-04:
I had this problem and it turned out that my phone had silently disconnected from the computer running the debugger so it was trying to run Flutter Blue Plus in an emulator, which doesn't work because the emulator doesn't have a Bluetooth adapter.

The solution was to unplug the phone from USB, plug it back in, then make sure to select the phone in the run/debug device selection dialog.

umarwaseemcowlar, 2023-06-20:
Maybe this helps

I encountered this problems while testing with `mocktio` and i had used a non mocked instance in my code instead of the mocked one