pauldemarco/flutter_blue #531: Fatal error after granting location permission

zathras, 2020-04-06:
After granting the app permission to access the location, I got this with version 0.7.1:

D/AndroidRuntime( 8228): Shutting down VM
E/AndroidRuntime( 8228): FATAL EXCEPTION: main
E/AndroidRuntime( 8228): Process: com.jovial.breezy_display, PID: 8228
E/AndroidRuntime( 8228): java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1452, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity {com.jovial.breezy_display/com.jovial.breezy.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Object io.flutter.plugin.common.MethodCall.arguments()' on a null object reference
E/AndroidRuntime( 8228): 	at android.app.ActivityThread.deliverResults(ActivityThread.java:4933)
E/AndroidRuntime( 8228): 	at android.app.ActivityThread.handleSendResult(ActivityThread.java:4976)
E/AndroidRuntime( 8228): 	at android.app.ActivityThread.access$1600(ActivityThread.java:222)
E/AndroidRuntime( 8228): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1849)
E/AndroidRuntime( 8228): 	at android.os.Handler.dispatchMessage(Handler.java:102)
E/AndroidRuntime( 8228): 	at android.os.Looper.loop(Looper.java:158)
E/AndroidRuntime( 8228): 	at android.app.ActivityThread.main(ActivityThread.java:7237)
E/AndroidRuntime( 8228): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 8228): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
E/AndroidRuntime( 8228): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
E/AndroidRuntime( 8228): Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Object io.flutter.plugin.common.MethodCall.arguments()' on a null object reference
E/AndroidRuntime( 8228): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.startScan(FlutterBluePlugin.java:740)
E/AndroidRuntime( 8228): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.onRequestPermissionsResult(FlutterBluePlugin.java:643)
E/AndroidRuntime( 8228): 	at io.flutter.embedding.engine.FlutterEnginePluginRegistry$FlutterEngineActivityPluginBinding.onRequestPermissionsResult(FlutterEnginePluginRegistry.java:612)
E/AndroidRuntime( 8228): 	at io.flutter.embedding.engine.FlutterEnginePluginRegistry.onRequestPermissionsResult(FlutterEnginePluginRegistry.java:356)
E/AndroidRuntime( 8228): 	at io.flutter.embedding.android.FlutterActivityAndFragmentDelegate.onRequestPermissionsResult(FlutterActivityAndFragmentDelegate.java:509)
E/AndroidRuntime( 8228): 	at io.flutter.embedding.android.FlutterActivity.onRequestPermissionsResult(FlutterActivity.java:611)
E/AndroidRuntime( 8228): 	at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:7263)
E/AndroidRuntime( 8228): 	at android.app.Activity.dispatchActivityResult(Activity.java:7141)
E/AndroidRuntime( 8228): 	at android.app.ActivityThread.deliverResults(ActivityThread.java:4929)
E/AndroidRuntime( 8228): 	... 9 more

dokinkon, 2020-04-17:
Got same issue.

d-silveira, 2020-04-24:
same here, it's a bit serious

dlangerm, 2020-04-27:
Same issue here

CodeMonkeyLin, 2020-04-28:
same issue here.need  relaunch the app

Alex-Evo, 2020-04-29:
Are you on Android 10? I think this is the reason:
https://developer.android.com/about/versions/10/privacy/changes#location-telephony-bluetooth-wifi

Google change the privacy requirements in Android 10 to require ACCESS_FINE_LOCATION for the following Bluetooth operations, but I think flutter_blue still only requests ACCESS_COARSE_LOCATION
BluetoothAdapter
- startDiscovery()
- startLeScan()
BluetoothAdapter.LeScanCallback
BluetoothLeScanner
- startScan()

Alex-Evo, 2020-04-29:
Here's a workaround that solved it for me for the time being:
Use [permission_handler](https://pub.dev/packages/permission_handler) to request the location permission (it requests both FINE and COARSE) before starting to interact with the FlutterBlue instance:
`await Permission.locationAlways.request().isGranted;`
(You can of course also use its return value to interact with the user in case the permission has not been granted...). This way you can ensure that the permission is already set correctly, and flutter_blue doesn't run into the problem.

MrCsabaToth, 2021-04-07:
Finally I ran into this, it results in a white screen and the app "not starting" just like the MissingPluginException or ProGuard type problems. I'll try https://pub.dev/packages/permission_handler

MrCsabaToth, 2021-05-01:
> Here's a workaround that solved it for me for the time being:
> Use [permission_handler](https://pub.dev/packages/permission_handler) to request the location permission (it requests both FINE and [...]

Where do you place this? Can I place it somewhere in the main or is that too early? (because it may induce some UI)
Or should I place it somewhere in an `initState`? Or in a `WidgetsBinding.instance.addPostFrameCallback` within an `initState`?

MrCsabaToth, 2021-05-02:
> Here's a workaround that solved it for me for the time being:
> Use [permission_handler](https://pub.dev/packages/permission_handler) to request the location permission (it requests both FINE and [...]

Another question after looking into `permission_handler`: `locationAlways` means background location on Q or above Android devices. In my case my app only presents "When in use" choice for the user. So I'd rather go with `locationWhenInUse` for a possible better outcome. What are your thoughts?