pauldemarco/flutter_blue #532: MissingPluginException (MissingPluginException(No implementation found for method setLogLevel on channel plugins.pauldemarco.com/flutter_blue/methods))

ramsestom, 2020-04-06:
Have this error when trying to run the example app on an android 5.1 device but this do not appear when on an android 8.0 device.
More details on this thread: https://github.com/pauldemarco/flutter_blue/issues/530 that its author has closed while the problem persists...

pauldemarco, 2020-04-06:
This is typically only seen in the emulator. Just to clarify, you are seeing this on an physical Android device? Can you share the model number?

ramsestom, 2020-04-06:
Yes I am seeing this on a physical Android device. This is a Xiaomi Redmi 3: https://www.gsmarena.com/xiaomi_redmi_3-7862.php with Android 5.1.1

SimeonVSimeonov, 2020-04-12:
And I seeing this too on a physical Android device. Huawei Y6II Compact : https://www.gsmarena.com/huawei_y6ii_compact-8663.php with Android 5.1.1

yageorge, 2020-09-24:
Same error on an old device HTC M8

GuylianStevens1, 2022-12-07:
Can the package be used while debugging with the Chrome extension. As I do to receive this same error?