pauldemarco/flutter_blue #534: How to get source device details like address & name?

AamirSk, 2020-04-07:
As of now, get the scanned result of devices where we get all details like id, name, etc. I want to get a source device id as soon as the source device starts Bluetooth. But I didn't find any API's to get such.

I tried with another wrapper package of this package that works fine with android but iOS not working.
Is there any convenient way to do so which should work on both Android & iOS?

ride4sun, 2020-04-07:
Here is how I do it. Unforturtunatly if you call `await device.discoverServices()` can only be called when you are connected to the device otherwise that method never comes back (don't see an exception). Hope that helps. 

```
Future _logDevices(List<ios.BluetoothDevice> devices,
      {bool discoverServices = true}) async {
    for (ios.BluetoothDevice device in devices) {
      await _logDevice(device, discoverServices: discoverServices);
    }
  }

  Future _logDevice(ios.BluetoothDevice device,
      {bool discoverServices = true}) async {
    _log.finest('Device name: ${device.name} Device id: ${device.id}');
    if (!discoverServices) return;
    List<ios.BluetoothService> deviceService = await device.discoverServices();
    _log.finest('Number of dicovered Services: ${deviceService.length}   ');
    for (var service in deviceService) {
      _log.finest('\tService Service: UUID: ${service.uuid}   ');
      for (var character in service.characteristics) {
        _log.finest('\t\tCharac.: UUID: ${character.uuid} ');
        _log.finest('\t\t\tCan read: ${character.properties.read} '
            'Can write: ${character.properties.write}  '
            'Can notify: ${character.properties.notify} ');
      }
    }
  }
```

pauldemarco, 2020-04-08:
Hi @AamirSk What are you referring to by the source device ID?
If it is the MAC Address, then this is not exposed by iOS for security reasons.

AamirSk, 2020-04-09:
> Hi @AamirSk What are you referring to by the source device ID?
> If it is the MAC Address, then this is not exposed by iOS for security reasons.

Yes  @pauldemarco, I was looking for MAC address of the self device(on which app is running). Actually, in quick analysis, I started with Android & I was getting expected Bluetooth ID & iOS was also giving search results of IDs. It was my mistake that I got to early conclusion & defined my use case considering I am getting correct ID's.

 With further study, I realized that iOS & some latest android devices don't share the exact Bluetooth ID of the device due to security reasons. So I need to think of different solutions.

AamirSk, 2020-04-09:
> Here is how I do it. Unforturtunatly if you call `await device.discoverServices()` can only be called when you are connected to the device otherwise that method never comes back (don't see an [...]

@ride4sun , Thank u so much for taking out ur time to respond to my query. Actually, I am looking for self device MAC ID along with scanned devices that don't look possible due to device security reasons.

ride4sun, 2020-04-09:
@pauldemarco  I think this can be closed.

closed 2020-04-09 by pauldemarco