pauldemarco/flutter_blue #535: dicoverService is broken

ride4sun, 2020-04-07:
@pauldemarco 
It seems like there is a long list of dicoverService() problems. I think the consent is that this needs at least a statement where you are with this and when these long-standing issues are fixed.
These are all open defects where many people desperately waiting for a solution or at least some shared thoughts. Any help or input on this would be great.

#509 #495 #426 #342 #276 #225

pauldemarco, 2020-04-08:
DiscoverServices should be called right after connecting to the device, typically in the BluetoothDeviceState handler after receiving the connected state.
I'm merging in some code to return an Exception when attempting this call on a device that's not connected.

pcourpron, 2020-04-26:
@pauldemarco 
Wouldn't it be better to be able to make this available before you connect to a device? Say if someone want to filter out a list of only approved device types for example.
Thanks!

endrelovas, 2020-05-24:
> @pauldemarco
> Wouldn't it be better to be able to make this available before you connect to a device? Say if someone want to filter out a list of only approved device types for example.
> Thanks!

That should be in AdvertismentData for you. If you are interested for other services (not advertised) you need to connect. Which makes the whole filter pointless as a connect/disconnect is costly.

So you can rely on these items within AdvertisementData:
- localName
- manufacturerData (if any)
- serviceUUIDs

If your device matches then you can interrogate it which need a connection AFAK. Correct me if I am wrong.

endrelovas, 2020-05-24:
discoverServices crashes my app without any error message. All i get is Lost connection to device. Exited (sigterm)

BayeNN, 2020-11-09:
@pauldemarco discoverServices issue mentioned in that #535 issue is reproduced on your example app as well (https://github.com/pauldemarco/flutter_blue/tree/master/example)
I'm using Android 9 (Pie)
The steps:
- start "flutter_blue_example" 
- scan for bt devices
- connect to a bt device - "BT device 001"
- hit "Back" 2 times to exit from app (app still running)
- go back to "flutter_blue_example"  - "BT device 001" is displayed as connected - hit Open
- hit "Refresh: icon (right to "Device is connected" text) - that code forces "device.discoverServices();" calling
- the exception occurs:

```
E/flutter (21021): [ERROR:flutter/lib/ui/ui_dart_state.cc(177)] Unhandled Exception: PlatformException(discover_services_error, no instance of BluetoothGatt, have you connected first?, java.lang.Exception: no instance of BluetoothGatt, have you connected first?
E/flutter (21021): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.locateGatt(FlutterBluePlugin.java:660)
E/flutter (21021): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.onMethodCall(FlutterBluePlugin.java:354)
E/flutter (21021): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:233)
E/flutter (21021): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:85)
E/flutter (21021): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:692)
E/flutter (21021): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter (21021): 	at android.os.MessageQueue.next(MessageQueue.java:326)
E/flutter (21021): 	at android.os.Looper.loop(Looper.java:181)
E/flutter (21021): 	at android.app.ActivityThread.main(ActivityThread.java:7073)
E/flutter (21021): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter (21021): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
E/flutter (21021): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)
```
I will really appreciate if you could provide us with some ideas how to fix / workaround this.

Thanks!