pauldemarco/flutter_blue #538: Scan Results running on BLE phone Moto 7g are always empty

ride4sun, 2020-04-09:
I am using ` flutter_blue: ^0.7.1+1` on flutter beta channel.
I am using a Moto 7g which is a BLE Phone and I don't get any devices in my scan result.
There is no Exception or Error message, no ideas where to go from here.
If I run the same program on IOS I see 3  BLE devices in my scan results.

pauldemarco, 2020-04-09:
Are you using any scan filters?

ride4sun, 2020-04-09:
The IOS and Android program is exactly the same. I do not even know how to set up a filter.

ride4sun, 2020-04-09:
Added #539 The permissions for Android where missing. @pauldemarco Is there a way to throw an exception when the permissions are missing

closed 2020-04-09 by ride4sun

RafaRuiz, 2022-03-15:
I still see no results and I'm still able to install the app from Google Play