pauldemarco/flutter_blue #539: Add Permission documentation.

ride4sun, 2020-04-09:
Can we add to the basic documentation:

**Add permissions for Bluetooth**
We need to add the permission to use Bluetooth and access location:

Android
In the AndroidManifest.xml let’s add:

 

```
> <uses-permission android:name="android.permission.BLUETOOTH" />
> <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
> <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
```

 

**iOS**
In the Info.plist let’s add:

 

```
<key>NSBluetoothAlwaysUsageDescription</key>
<string>Need BLE permission</string>
<key>NSBluetoothPeripheralUsageDescription</key>
<string>Need BLE permission</string>
<key>NSLocationAlwaysAndWhenInUseUsageDescription</key>
<string>Need Location permission</string>
<key>NSLocationAlwaysUsageDescription</key>
<string>Need Location permission</string>
<key>NSLocationWhenInUseUsageDescription</key>
<string>Need Location permission</string>
 
```

For location permissions on iOS see more at https://developer.apple.com/documentation/corelocation/requesting_authorization_for_location_services

davebaraka, 2020-05-02:
Why is location needed to access bluetooth?

boskokg, 2020-05-02:
https://developer.android.com/guide/topics/connectivity/bluetooth-le#permissions

davebaraka, 2020-05-02:
According to [this](https://www.polidea.com/blog/a-curious-relationship-android-ble-and-location/) article, it seems like location is not necessary for iOS unless beacon advertisements are used.

pedrobagatin, 2020-05-02:
It would be nice if this step was explained in the "installing" section

ride4sun, 2020-05-02:
That why I created that issue. I think a lot of questions/issues for this lib could be avoided if there would be a better documentation.