pauldemarco/flutter_blue #540: Throw Exception in Android and IOS when the Permissions necessary are missing

ride4sun, 2020-04-09:
It looks like that would be possible and helps new people to waste time.
related to #539

hwesley, 2020-04-17:
I'm only using this library on iOS at this moment. I think it would be better to have specific methods to check if permission is granted or not. At this moment the isOn method both checks the permission and if bluetooth is enabled or not. Which makes it impossible to differentiate & notify the user correctly. 

Regarding the iOS permission, i'm also wondering if it's possible to not trigger it once the app starts up, but trigger it with a method, only when you want it to trigger.

ride4sun, 2020-04-17:
There is actually a lib for that:
permission_handler: ^4.2.0+hotfix.3

Alex-Evo, 2020-04-29:
@ride4sun permission_handler only works for non-Bluetooth permissions, see here: https://github.com/Baseflow/flutter-permission-handler/issues/177

Also note that Android 10 requires ACCESS_FINE_LOCATION instead of ACCESS_COARSE_LOCATION.

Regarding the Bluetooth permissions, I also think that delaying the initialization of BluetoothCoreManager so that it can be triggered at a well-defined point in the App workflow would make much more sense than permission request pop up uncommented at App start.

qimingweng, 2020-06-11:
I noticed there's a PR for not requesting permissions until the app actually uses BLE: https://github.com/pauldemarco/flutter_blue/pull/599