pauldemarco/flutter_blue #543: Conflict with Firebase_messaging with Plugin

smlertek, 2020-04-10:
when we implement firebase_messaging and we include specific this 3 lines :

InheritedApp.of(context).appGlobal.firebaseMessagingConfig();
InheritedApp.of(context).appGlobal.notification.notificationConfig(onSelectNotification);
InheritedApp.of(context).appGlobal.notification.setNotificationSpecifics();

flutter_blue gives us this error and doesn´t start scanning:

`I/MediaCodec(13427): Codec shutdown complete
D/BluetoothAdapter(13427): STATE_ON
D/BluetoothAdapter(13427): STATE_ON
D/BluetoothAdapter(13427): STATE_ON
D/BluetoothAdapter(13427): STATE_ON
D/BluetoothLeScanner(13427): could not find callback wrapper
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): Failed to handle method call
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): java.lang.NullPointerException: Attempt to invoke virtual method 'int android.content.Context.checkPermission(java.lang.String, int, int)' on a null object reference
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at androidx.core.content.ContextCompat.checkSelfPermission(ContextCompat.java:535)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.onMethodCall(FlutterBluePlugin.java:241)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:231)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:93)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:642)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at android.os.MessageQueue.next(MessageQueue.java:323)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at android.os.Looper.loop(Looper.java:136)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at android.app.ActivityThread.main(ActivityThread.java:6682)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at java.lang.reflect.Method.invoke(Native Method)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1520)
E/MethodChannel#plugins.pauldemarco.com/flutter_blue/methods(13427): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1410)
I/flutter (13427): Error starting scan.
E/flutter (13427): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] **Unhandled Exception: PlatformException(error, Attempt to invoke virtual method 'int android.content.Context.checkPermission(java.lang.String, int, int)' on a null object reference, null)**
E/flutter (13427): #0      FlutterBlue.scan 
package:flutter_blue/src/flutter_blue.dart:103
E/flutter (13427): <asynchronous suspension>
E/flutter (13427): #1      FlutterBlue.startScan 
package:flutter_blue/src/flutter_blue.dart:132
E/flutter (13427): #2      _MainPageState.scanDevices 
package:covid_radar/…/ui/mainPage.dart:255
E/flutter (13427): #3      _MainPageState.initPlatformState.<anonymous closure> 
package:covid_radar/…/ui/mainPage.dart:190
E/flutter (13427): #4      _rootRunUnary  (dart:async/zone.dart:1134:38)
E/flutter (13427): #5      _CustomZone.runUnary  (dart:async/zone.dart:1031:19)
E/flutter (13427): #6      _CustomZone.runUnaryGuarded  (dart:async/zone.dart:933:7)
E/flutter (13427): #7      _CustomZone.bindUnaryCallbackGuarded.<anonymous closure>  (dart:async/zone.dart:970:26)
E/flutter (13427): #8      _rootRunUnary  (dart:async/zone.dart:1138:13)
E/flutter (13427): #9      _CustomZone.runUnary  (dart:async/zone.dart:1031:19)
E/flutter (13427): #10     _CustomZone.bindUnaryCallback.<anonymous closure>  (dart:async/zone.dart:954:26)
E/flutter (13427): #11     _Timer._runTimers  (dart:isolate-patch/timer_impl.dart:384:19)
E/flutter (13427): #12     _Timer._handleMessage  (dart:isolate-patch/timer_impl.dart:418:5)
E/flutter (13427): #13     _RawReceivePortImpl._handleMessage  (dart:isolate-patch/isolate_patch.dart:174:12)
E/flutter (13427):`

thermoChas, 2020-05-01:
Did you find an answer for this? I am seeing the same thing.

ened, 2020-05-05:
It's because firebase_messaging registers a background isolate without a activity. That one calls "registerWith" in the flutter_blue plugin code, which overrides a *static* instance variable, thus eliminating the previously held activity reference.

@pauldemarco are you working on a refactor for this, yet? I'd like to submit a PR..

adamfraczkowski, 2020-05-26:
I had the same problem after add flutter_core into project. Did you resolve this issue ? How to make bluetooth scan working ? @ened

ened, 2020-05-27:
@adamfraczkowski, you can try to adjust your pubspec.yaml and point to my fork with the single fix:

```
dependency_overrides:
  flutter_blue:
    git:
      url: https://github.com/ened/flutter_blue.git
      ref: 98fca2b4c051e3fafe297f9e7dca18a409ff2130
```

kanhaic, 2020-07-07:
I am having this same issue, How did you resolve this. I am still getting this error.