pauldemarco/flutter_blue #544: Not retrieving bluetooth mobile devices

vinay-khatri, 2020-04-10:
only finds devices like my tv, laptop, external desktop adapter, mi band. But no mobile device ever. plz provide setting to scan mobile devices too. Thx

vinay-khatri, 2020-04-11:
You can do this by startDiscovery() in android. and listen to broadcast receiver.
It returns all classic Bluetooth devices.

ride4sun, 2020-04-14:
I see a bunch of devices but only the once which I have names are BLE devices. At least that is what I think. The documentation of this lib is rather thin.

```
utter: ------- Scan result: 9 ------------
2020-04-13 16:29:26.131 28107-28161/com.cobd.chkobd.debug I/flutter: Device name: - no Name - Device id: 19:91:E9:35:91:CA
2020-04-13 16:29:26.133 28107-28161/com.cobd.chkobd.debug I/flutter: Device name: - no Name - Device id: 74:C0:56:C6:C5:98
2020-04-13 16:29:26.136 28107-28161/com.cobd.chkobd.debug I/flutter: Device name: VEEPEAK Device id: 34:81:F4:CE:33:05
2020-04-13 16:29:26.139 28107-28161/com.cobd.chkobd.debug I/flutter: Device name: - no Name - Device id: 51:97:50:CD:57:4C
2020-04-13 16:29:26.142 28107-28161/com.cobd.chkobd.debug I/flutter: Device name: - no Name - Device id: 55:AC:23:23:EC:0F
2020-04-13 16:29:26.144 28107-28161/com.cobd.chkobd.debug I/flutter: Device name: - no Name - Device id: 54:14:14:D2:E8:11
2020-04-13 16:29:26.148 28107-28161/com.cobd.chkobd.debug I/flutter: Device name: - no Name - Device id: 0B:B4:47:83:3B:80
2020-04-13 16:29:26.151 28107-28161/com.cobd.chkobd.debug I/flutter: Device name: - no Name - Device id: 7E:CC:65:4D:4A:18
2020-04-13 16:29:26.153 28107-28161/com.cobd.chkobd.debug I/flutter: Device name: Charge 3 Device id: C1:FC:78:AA:52:05
```

skabirgithub, 2020-04-25:
I am also facing same problem. The example app can not list other phones with bluetooth but can get my laptop in the list. Please helpme to fix out