pauldemarco/flutter_blue #547: Dont get this plugin to work

ride4sun, 2020-04-14:
I wasted tons of time with this plugin and I am wondering if somebody is successfully using this plugin for both platforms?

If yes I would like to hire somebody to consult with me for a few hours to work my kinks out. 
I can not even get the examples to successfully work. The software I am writing is successfully using flutter_bluetooth_serial plugin without any problems and would like to support both platforms on ble as well.

If somebody considers that please contact me here or at ride4sun@gmail.com. I speak fluent English and German. Screensharing and Audio would be great.

ajayrathode, 2020-04-17:
Hey @ride4sun , i am using flutter_blue to get list of bluetooth devices and their rssi value. But i am only getting my TV listed. My android Phones,iphone is not getting listed in bluetooth scan. 

Does this flutter_blue works only with BLE devices and not mobile phones ??
Do you have results other than mine? If yes, Please let me know.

andrewknockin, 2020-04-18:
I also couldn't get the example to work. I have successfully got part of mine working now. It scans and runs on my iPhone6. It can see other iPhones and a device I am trying to communicate with. It also sees my Samsung TV. I am disappointed that there is no good working example available even elsewhere on github. All the other github examples are either direct copies of example and not working, or don't even compile (Laurie Marceau). The ReadMe appears to have good instructions, I just don't know where to put them in my app.

ajayrathode, 2020-04-25:
Hi Andrew,

Could you please drop your code snippet for achieving above said
functionality. I am new to flutter and looking for something to start with
on this . Troed myself but didnt worked for me. In case you want, drop me a
mail at vaddiajayrathode@gmail.com

andrewknockin, 2020-04-25:
I have put it on public github at https://github.com/andrewknockin/flutter_blue_lib.