pauldemarco/flutter_blue #553: Characteristic notifications are too slow in the specific case.

lunakillz, 2020-04-18:
First of all, I am using your library very well.
Thank you very much. :)

When I received Characteristic notifications at a very quickly(ex: 10ms ). 
Application had become very slow at Release mode. 

Because "**GUID._fromString**" is too slow at **AOT**...
 **RegExp** that is used in "_fromString" is slower at "**AOT**" than when "**JIT**"

[RegExp is KnownIssue](https://github.com/dart-lang/sdk/issues/23249)

In my case, I do not use RegExp
I use String.codeUnit to validate a GUID.

**Please consider excluding  RegExp.**

pauldemarco, 2020-04-18:
I just merged an excellent PR from Enrico (@ezamagni) that addresses this: https://github.com/pauldemarco/flutter_blue/pull/552

Please try targeting flutter_blue master branch in your project and let me know if that helps.

pauldemarco, 2020-04-20:
Flutter version 0.7.2 was published with the performance fix included. @lunakillz Please try out this version and close the issue if resolved.

Thanks.

lunakillz, 2020-04-21:
Sorry for late response

0.7.2 works well and quickly!!!
It's amazing that Enrico (@ezzamagni) approached it in a similar way as I did.

Thanks all

closed 2020-04-21 by lunakillz