pauldemarco/flutter_blue #554: Flutter_blue issues minSdkVersion 19

pixelatedpic, 2020-04-20:
My current build settings

android {
    compileSdkVersion 28

    defaultConfig {
        minSdkVersion 19
        targetSdkVersion 28

the flutter_blue version tried

  flutter_blue: ^0.7.1+1
  flutter_blue: ^0.6.3+1

either of this worked.

This is the debug message

###########################################

Launching lib\main.dart on Android SDK built for x86 in debug mode...
[!] Your app isn't using AndroidX.
    To avoid potential build failures, you can quickly migrate your app by following the steps on https://goo.gl/CP92wY.
D:\flutter_Stuffs\bleapps\android\app\src\debug\AndroidManifest.xml Error:
	uses-sdk:minSdkVersion 16 cannot be smaller than version 19 declared in library [:flutter_blue] D:\flutter_Stuffs\bleapps\build\flutter_blue\intermediates\merged_manifests\debug\processDebugManifest\merged\AndroidManifest.xml as the library might be using APIs not available in 16
	Suggestion: use a compatible library with a minSdk of at most 16,
		or increase this project's minSdk version to at least 19,
		or use tools:overrideLibrary="com.pauldemarco.flutter_blue" to force usage (may lead to runtime failures)

FAILURE: Build failed with an exception.
* What went wrong:
Execution failed for task ':app:processDebugManifest'.
> Manifest merger failed : uses-sdk:minSdkVersion 16 cannot be smaller than version 19 declared in library [:flutter_blue] D:\flutter_Stuffs\bleapps\build\flutter_blue\intermediates\merged_manifests\debug\processDebugManifest\merged\AndroidManifest.xml as the library might be using APIs not available in 16
  	Suggestion: use a compatible library with a minSdk of at most 16,
  		or increase this project's minSdk version to at least 19,
  		or use tools:overrideLibrary="com.pauldemarco.flutter_blue" to force usage (may lead to runtime failures)
* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org
BUILD FAILED in 1m 47s
Exception: Gradle task assembleDebug failed with exit code 1
Exited (sigterm)

###########################################

pauldemarco, 2020-04-20:
@pixelatedpic Android Bluetooth requires a minimum SDK of 19, so therefore flutter_blue plugin will require that of your project as well.

You can set your project's minSdkVersion to 19 in the file `bleapps\android\app\build.gradle`.
Example:
```
defaultConfig {
        minSdkVersion 19
    }
```

closed 2020-04-20 by pauldemarco

pixelatedpic, 2020-04-29:
thanks @pauldemarco for support