pauldemarco/flutter_blue #556: Example could not scan any bluetooth device?

newbieflutter, 2020-04-24:
Hi Paul,
            I have tested you latest example but it seems not to be scanning any of those devices? Is there any thing I am missing?

Wassim24, 2020-05-12:
@newbieflutter try enabling the Location. It solved the problem for me.

spekary, 2020-11-26:
To expand on this, as I ran into this issue on Android 10.

I had ACCESS_FINE_LOCATION in the manifest. Upon launching the app, the OS asked to approve accessing the location, which I did. Calling FlutterBlue.instance.isAvailable and isOn both returned true. However, scanning appeared to freeze. Scanning started, but just never reported anything back in the stream.

The issue was that I had "Location" services turned off in the preferences of the Android device itself. I noticed that Nordic's nrfConnect app actually had a banner when it started saying that Location was off.

So, it looks like FlutterBlue should do an additional check for this before reporting true in isAvailable, or perhaps it should return an exception when calling "scan". Yes, Bluetooth was On, but scanning was not going to work without the Location preference.

I suppose this is kind of tricky. I imagine that connecting with a previously paired device will work fine with the Location preference off, hence isAvailable and isOn would be true.

Other options are to have an additional flag "canScan" that would check this. Or, the easiest, add it to the ReadMe file and document it in the scan routine that users of FlutterBlue should check for this before calling Scan. Not sure what the correct cross-platform way to do this should be.

@pauldemarco , do you have any thoughts on what the right thing to do here would be? It is really nice that flutter blue is cross-platform, but this is really a very specific issue on recent versions of Android.