pauldemarco/flutter_blue #557: [iOS] [13.4] Can't connect to device after power cycle

gurgenDP, 2020-04-26:
**flutter_blue: ^0.6.3+1**
**.connect()** doesn't workd after updating the phone to iOS 13.4. 

**.connect()** just doesn't work and doesn't throw any error, even the timeout is not throwing an error. 

Works just fine on 13.3.1 and on 13.4 when forgetting the device from iOS Settings. 

Connection snippet:
```dart
      // Check if already connected
      if (_status != BluetoothDeviceState.connected) {
        // Connect to device
        try {
          await device
              .connect(
                  timeout: Duration(seconds: _connectionTimeout),
                  autoConnect: true)
              .catchError((e) {
            print('ERROR WHILE CONNECTING TO DEVICE + #$e');
            completer.complete(false);
          });
        } catch (e) {
          print('ERROR WHILE CONNECTING TO DEVICE');
          completer.complete(false);
        }
      }
```

When trying to connect with LightBlue it fails with this error on the logs
```
(Error: Peer removed pairing information)
```

Not sure if the plugin should handle this and initiate a new connection, but for sure it must throw an error.

francescomapelli, 2020-05-30:
I am probably experiencing a similar issue: 
1) the first time the connection works
2) after disconnect and after we reboot the bluetooth device it is not possible to reconnect
3) removing the device from the ios settings allow me to reconnect but we are back to step 1

gurgenDP, 2020-05-31:
@francescomapelli As a workaround, we end up with getting rid of bonding at all. Connection performance is about the same, with no negative impact in our case.

francescomapelli, 2020-05-31:
@gurgenDP that may work! how were you able to avoid bonding?

gurgenDP, 2020-05-31:
@francescomapelli I should mention that we did it on the LE device side. I cannot tell you exact way, because it's depending on BLE module. Generally it's specific setting and should be not hard to find.

francescomapelli, 2020-05-31:
@gurgenDP Thanks! We'll look into it :)

francescomapelli, 2020-05-31:
Just  a last question: you mention you had to forget the device from the ios settings to have it working again. Now, without bonding, do you still have the device in the settings (but you do not need to remove it anymore for it to reconnect) or you are not seeing it in the settings at all?

gurgenDP, 2020-06-01:
I don't have it's on the device settings list. Not sure 100%, but those are bonded and paired devices only.