pauldemarco/flutter_blue #558: Scanresult device id on iOS shows a long ID

KevinRohn, 2020-04-27:
Hello community,

when I check the scan results on iOS it just shows me a long id and on android, it shows me the correct mac address.
Is there a way to show the MAC address on iOS?

Thank you in advance,

cheers
Kevin

keehoo, 2020-05-06:
I'm experiencing the same thing also, the name of the device is different from Android. I would really appreciate someone having a look at this?

`flutter_blue: ^0.7.2`

`child: Text(
                '${device.id.id.toUpperCase()} ${device.name.toUpperCase()}',
                style: TextStyle(color: Colors.white),
              ),`

[image]

keehoo, 2020-05-07:
Just to add to this request: I found the following snippets https://stackoverflow.com/questions/18973098/get-mac-address-of-bluetooth-low-energy-peripheral

please take a look at jlandyr's answer (answered Aug 29 '19 at 11:11)

Perhaps someone is capable of answering if this is even possible to do?

fyredave, 2020-08-13:
This is also causing me issues, in my case I'd like to identify if it's one of our IoT devices using the scanresults but this doesn't seem possible on iOS without checking the characteristics

jkano, 2020-11-22:
This is a limitation on iOS, not an issue of flutter_blue. This is the way as iOS identifies a device. It computes a unique identifier (uuid) using the device mac but the mac itself is not available when scanning a Bluetooth device.

KevinRohn, 2020-11-23:
> This is a limitation on iOS, not an issue of flutter_blue. This is the way as iOS identifies a device. It computes a unique identifier (uuid) using the device mac but the mac itself is not available when scanning a Bluetooth device.

Thank you for the answer. I've also checked it in the apple developer forum and I seems to be correct. It seems like it'a an security design to not show the MAC address, because it can also be changed. https://developer.apple.com/forums/thread/105064

closed 2020-11-23 by KevinRohn