pauldemarco/flutter_blue #563: Listing nearby phones that have bluetooth on

kalinkochnev, 2020-04-29:
So I'm trying to get a list all bluetooth devices (phones specifically) but Flutter doesn't seem to be seeing the phones.
I have this method called on a button press just to test what gets output. Here is the code for it

```dart
//Listen to scan results
flutterBlue.startScan(timeout: Duration(seconds: 4));

var subscription = flutterBlue.scanResults.listen((scanResult) {
  for (ScanResult scan in scanResult) {
    BluetoothDevice device = scan.device;
    print('${device.name} found! rssi: ${scan.rssi}dBm');
  }
});
```
After pressing the button I get this:

    I/flutter (12864): Device1 found! rssi: -94dBm
    I/flutter (12864): Device1 found! rssi: -94dBm
    I/flutter (12864):  found! rssi: -70dBm
    I/flutter (12864): Device1 found! rssi: -94dBm
    I/flutter (12864):  found! rssi: -70dBm
    I/flutter (12864): Jabra Evolve 75e found! rssi: -70dBm
    I/flutter (12864): Device1 found! rssi: -94dBm
    I/flutter (12864):  found! rssi: -70dBm
    I/flutter (12864): Jabra Evolve 75e found! rssi: -70dBm
    I/flutter (12864): Inspire HR found! rssi: -86dBm

Some weird things are happening. I seem to be getting duplicate listings and there seems to be a device with no name. But the phone I enabled discoverability on does not show up. I'm expecting something like this:  `Moto G5 found! rssi:##dBm`

I also connected directly to the phone and tried to list all connected devices with no luck:
```dart
for (BluetoothDevice device in await flutterBlue.connectedDevices) {
    print('Connected device: ${device.name} ${device.id}');
});
```
Output:
```
D/FlutterBluePlugin(12864): mDevices size: 0
D/FlutterBluePlugin(12864): mDevices size: 0
I/Surface (12864): opservice is null false
```

Does flutter_blue not support scanning for phones? I just started using this so I'm not sure what I'm doing. 

Thanks for the help!

bilaalabdelhassan, 2020-06-07:
I just came here for the same problem.

I can find my PC, TV, even by Samsung Watch but it can't detect a single phone.

I really hope there is a solution for this.

Akilesh30, 2020-08-06:
You guys got any solution?

kalinkochnev, 2020-08-06:
I put up a stack overflow post about this a while ago. I haven't tried this but it seems promising: https://stackoverflow.com/questions/61492642/listing-nearby-phones-with-bluetooth-on-flutter-blue-package/62351551#62351551