pauldemarco/flutter_blue #566: ios13  gray screen;

cander0815, 2020-05-05:
Obtaining permissions when entering the app for the first time on ios13.3.2 will cause the app to get stuck on a gray screen;

FaresKis, 2020-05-05:
Same problem here : 
`2020-05-05 17:21:33.497289+0200 Runner[22288:6622190] [VERBOSE-2:engine.cc(127)] Engine run configuration was invalid.
2020-05-05 17:21:33.497366+0200 Runner[22288:6622190] [VERBOSE-2:shell.cc(437)] Could not launch engine with configuration.`

At first I had an issue with permissions, so I added them on

cander0815, 2020-05-07:
> Same problem here :
> `2020-05-05 17:21:33.497289+0200 Runner[22288:6622190] [VERBOSE-2:engine.cc(127)] Engine run configuration was invalid. 2020-05-05 17:21:33.497366+0200 Runner[22288:6622190] [...]

Hello, my problem confirmation is caused by the configuration of multiple languages, it has been fixed, you confirm the cause of your problem, or re-issue a problem

closed 2020-05-07 by cander0815