pauldemarco/flutter_blue #569: Connecting to a device without scan

slecoustre, 2020-05-06:
Is it possible to connect to a device without having to scan?
I have the address of the device.

Thank you

boskokg, 2020-05-06:
You can, but only on Android

slecoustre, 2020-05-07:
OK, I can't find the method in the package in Device or FlutterBlue.

Can you tell me how to find it?

boskokg, 2020-05-07:
I have done it this way:      
      protos.BluetoothDevice dev = protos.BluetoothDevice()..remoteId = address;
      var _gattClient = BluetoothDevice.fromProto(dev);
      var _gattClient.connect(autoConnect: everConnectedDevicesAddresses.contains(address));

Note: be carefull with autoConnect option. I just fixed a big issue on LG phones... I could not autoconnect after bluetooth reset. So I made some Set of connected devices, that I am clearing when bluetooth is turning off or turned off. autoconnect parameter should be true only if device was connected at least once after bluetooth or phone reset.