pauldemarco/flutter_blue #575: Impossible to set log level

Lorenzo-Venturelli, 2020-05-10:
Hi,
I'm having problems when setting the log level for FlutterBlue plugin. I have inplemented my code as the following:

federicocaroli, 2020-05-10:
I know hundreds of people that have the same problem.

ericmichael, 2020-05-24:
By any chance did you all figure this out? @ride4sun  @federicocaroli

ride4sun, 2020-05-25:
I did I switched to another Bluetooth implementation .
https://pub.dev/packages/flutter_ble_lib. Works for IOS and Android BLE.
All my problems basicly were gone. I abstracted the Bluetooth already
before and I had all working in a few days. The log level problem is
related to flutter ble code logging to the highest level. Therefore you can
not disable it. It is a easy fix and a long time known and reported. It
just has to be fixed.

Annelein, 2021-04-16:
is there still no way to solve this?

martinb741, 2021-05-05:
I was really annoyed by this issue, too, and had a look at the flutter_blue code. It seems that the log level is set to debug internally once the device becomes available. If we use setLogLevel() immediately after getting the instance then our log level is actually set correctly before the device becomes available but then it gets overwritten by the internal call setting it to debug.

As a workaround we can just wait until the device is available before setting the log level. For example this works well for me:

```
FlutterBlue _flutterBlue = FlutterBlue.instance;
_flutterBlue.isAvailable.then((isAvailable) {
  if (isAvailable) _flutterBlue.setLogLevel(LogLevel.info);
});
```

For fixing the problem in flutter_blue I can think of a few ways
- Do not set the initial default log level in flutter_blue.dart at all, but rather in the code of the underlying platforms
- Skip the call to _setLogLevelIfAvailable() and directly call setLogLevel() internally when setting the initial log level. I am not entirely sure, but I think this should be ok, because we can call setLogLevel() immediately after getting the instance and it works - it just gets overwritten afterwards.
- Check for isAvailable inside setLogLevel(). That way we can also skip the call to _setLogLevelIfAvailable()

iJamesPHP, 2021-07-23:
> I was really annoyed by this issue, too, and had a look at the flutter_blue code. It seems that the log level is set to debug internally once the device becomes available. If we use setLogLevel() [...]

I was hopeful after reading this, but it didn't work for me :( I am still constantly seeing all of the debug from FlutterBlue. I do a lot of constant scanning/connecting so this logging issue is really getting in my way. I don't want to migrate to another package and I don't want to alter the source code because I will always have to update the package again.

gumbypp, 2021-07-23:
I ran into this issue too. I eventually decide to write a post-install script that I run in my build process after `flutter pub cache repair`:

```
#!/bin/bash
#
# a) The Flutter Blue library currently uses NSLog() to write to console on a running device. 
#    For security reasons, we want to supress those. So be sure to run this script before
#    doing a release build using "flutter build ios".
#
# b) Sometimes builds yield the following error:
#    [iOS] fatal error: 'flutter_blue/FlutterBluePlugin.h' file not found
#    This is fixed by editing flutter_blue.podspec per https://github.com/pauldemarco/flutter_blue/issues/386

sed -i '.bak' 's/ NSLog/\/\/NSLog/g' /Users/$USER/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.2/ios/Classes/FlutterBluePlugin.m

sed -i '.bak' 's/ss\.header_mappings_dir/#ss\.header_mappings_dir/g' /Users/$USER/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.2/ios/flutter_blue.podspec
```

This disables all logging for iOS. You could probably do something similar for Android. Yes, it alters the source code, but might be easier than migrating to another package.