pauldemarco/flutter_blue #577: Finding BluetoothState.unknown on iOS

RafaRuiz, 2020-05-12:
Hello.

I have an app that shows the BluetoothState in the first screen.
It's weird, but most of the times (60%) when the app comes from TestFlight (and a few times with a normal build, I would say 2%) when I start the app, I receive the `BluetoothState.unknown` on **iOS**.
I have the last version, but it's quite curious. I've tried to find this issue, if anyone had it, but I can't find it.
It's quite difficult to recreate because I have to constantly kill the app and open it again to be lucky and recreate it, and the only good scenario to recreate it is on TestFlight.

What could be causing it?

I've checked and the permissions are ok (when they're not, the state is `BluetoothState.unauthorized`).

RafaRuiz, 2020-05-13:
A nice update.
I've found out that this issue happens only when you tap the app from TestFlight directly from your gallery of apps.
If you tap TestFlight > [App] > "OPEN", it doesn't happen!

Not sure if now this is a TestFlight error or an issue with the library itself being in a nutshell.

RafaRuiz, 2020-05-20:
Was due to leaving a static reference of its state.
Bad boy @RafaRuiz .

closed 2020-05-20 by RafaRuiz

debugerer, 2020-06-21:
I encounter the same issue when  normal boot the example app without TestFlight.

RafaRuiz, 2020-06-22:
@debugerer I don't know how to help without any more details. My case wasn't the example app, it was more overengineered and I left static Streams with the BluetoothState

debugerer, 2020-06-27:
Just run the example app,  you will reproduce the bug more than 50% .