pauldemarco/flutter_blue #580: Unable to get both bluetooth device list (paired and near by) in Android version 9

rahul-jain-7893, 2020-05-14:
By using this code. Unable to get bluetooth device list.

> flutterBlue.startScan(timeout: Duration(seconds: 4));
> 
> // Listen to scan results
> var subscription = flutterBlue.scanResults.listen((results) {
>     // do something with scan results
>     for (ScanResult r in results) {
>         print('${r.device.name} found! rssi: ${r.rssi}');
>     }
> });
> 
> // Stop scanning
> flutterBlue.stopScan();

AuspeXeu, 2020-05-14:
Same for me under Android 9, it works sometimes though...

Darren-FanDemand, 2020-05-18:
I tried on both android 9 and 6. I don't get the list. Please assist.

Hritik14, 2020-05-18:
Could be related to https://github.com/pauldemarco/flutter_blue/issues/581

AuspeXeu, 2020-05-18:
At least not for me, I have that in my manifest :)

Darren-FanDemand, 2020-05-19:
I added the permission but that didn't seem to make a difference. I'm testing from an android phone. I have two additional devices with bluetooth switched on (another android phone and an iPod). They all see each other via the OS. When I run the app, the plugin does not see those devices. Funny enough though, today it consistently detected a random LG device somewhere in my building but not the devices I have right here in my apartment.

Hritik14, 2020-05-19:
That is weird. Can you try installing the example file that comes along?

Hritik14, 2020-05-19:
I can confirm this issue. 
The example app is unable to identify some of the devices nearby and is able to detect others.
Device detected: `08:25:25:7a:XX:XX`
Device undetected: `38:e6:0a:e3:XX:XX`

AuspeXeu, 2020-05-19:
Just a hunch maybe it's related to the `allowDuplicates` option? Also, I just realized the code from the initial issue stops scanning immediately again...

philos3, 2020-05-20:
> By using this code. Unable to get bluetooth device list.

add uses-permission in androidMainifest.xml

   <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />

    <uses-feature
        android:name="android.hardware.bluetooth_le"
        android:required="true" />

Darren-FanDemand, 2020-05-20:
> > By using this code. Unable to get bluetooth device list.
> > > flutterBlue.startScan(timeout: Duration(seconds: 4));
> > > // Listen to scan results
> > > var subscription = flutterBlue.scanResults.listen((results) {
> > > // do something with scan results
> > > for (ScanResult r in results) {
> > > print('${r.device.name} found! rssi: ${r.rssi}');
> > > }
> > > });
> > > // Stop scanning
> > > flutterBlue.stopScan();
> 
> add uses-permission in androidMainifest.xml
> 
> ```
> <uses-permission android:name="android.permission.BLUETOOTH" 
> />
>     <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" 
> />
>     <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"
>  />
>  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
>     <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"
> />
> 
>     <uses-feature
>         android:name="android.hardware.bluetooth_le"
>         android:required="true" />
> ```

Thanks but this doesn't solve the current problem. The plugin still doesn't detect all the devices around.

Wassim24, 2020-05-25:
Have you tried enabling location ? I had the same exact problem once, this solved it.
Also, flutter blue supports only BLE (Beacons) if i understand correctly, it won't detect others.

AuspeXeu, 2020-05-28:
Just some clues I found. I have the same BT device one time paired to my Pixel XL, one time not paired. When paired, I don't find it. When not paired, I find it. Did the same experiment with a Samsung S9. For this one it's the opposite. When paired, I find it. When not paired, I don't find it.

Can anyone make sense of this?

zwells, 2020-06-19:
Maybe add the following in AndroidManifest.xml:

<uses-permission android:name="android.permission.FOREGROUND_SERVICE" />

jianboy, 2021-12-11:
It only can scan PC(windows 10) bluetooth device.

**But android device can not find.** Why?