pauldemarco/flutter_blue #583: discoverServices crashes in "toDescriptorProto" on iOS

endrelovas, 2020-05-25:
FlutterBluePlugin.m crashes on line 710:

`int value = [descriptor.value intValue];`
 
Error message:

> 2020-05-25 10:55:57.601017+0200 Runner[4579:3690495] -[_NSInlineData intValue]: unrecognized selector sent to instance 0x280255100
> 2020-05-25 10:55:57.601449+0200 Runner[4579:3690495] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[_NSInlineData intValue]: unrecognized selector sent to instance 0x280255100'

I use a custom version of 0.7.0 due to the unfixed empty notification error in flutter_blue in the repo.

philpeterman, 2020-09-18:
I had a very similar issue but wanted to return the values in a specific format.  Here is my take on this:

`Changes are at line 716 and 507
 
ProtosReadDescriptorResponse *result = [[ProtosReadDescriptorResponse alloc] init];
  [result setRequest:q];
    if([descriptor.value isKindOfClass:[NSNumber class]]){
        value = [descriptor.value intValue];
    }
    if([descriptor.value isKindOfClass:[NSData class]]){
        value = CFSwapInt16BigToHost(*(int*)([descriptor.value bytes]));
    }
  [result setValue:[NSData dataWithBytes:&value length:sizeof(value)]];
  [_channel invokeMethod:@"ReadDescriptorResponse" arguments:[self toFlutterData:result]];`

Jmstewart1127, 2021-01-09:
> FlutterBluePlugin.m crashes on line 710:
> 
> `int value = [descriptor.value intValue];`
> 
> Wrapping it inside a @Try @catch clause eliminates the crash.
> 
> I beleive it is a wrong assumption that only int values returned.
> 
> Error message:
> 
> > 2020-05-25 10:55:57.601017+0200 Runner[4579:3690495] -[_NSInlineData intValue]: unrecognized selector sent to instance 0x280255100
> > 2020-05-25 10:55:57.601449+0200 Runner[4579:3690495] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[_NSInlineData intValue]: unrecognized selector sent to instance 0x280255100'
> 
> I use a custom version of 0.7.0 due to the unfixed empty notification error in flutter_blue in the repo.
> 
> **EDIT**
> 
> I replaced the code. Seems to work fine.
> 
> ```
>   @try {
>         if  ([descriptor.value isKindOfClass:NSNumber.class]) {
>             int value = [descriptor.value intValue];
>             [result setValue:[NSData dataWithBytes:&value length:sizeof(value)]];
>         } else {
>             [result setValue:descriptor.value];
>         }
>     } @catch (NSException *exception) {
>         NSLog(exception.reason);
>     }
> ```

This worked for me on 0.7.3. However, I had to replace `int value = [descriptor.value intValue];` on lines 509 and 712. It would be great to get this or a similar fix merged in.

alexflav23, 2021-02-13:
The correct working and tested fix is:

```
  NSData *data = descriptor.value;
  NSUInteger len = [data length];
  Byte *byteData = (Byte*)malloc(len);
  memcpy(byteData, [data bytes], len);

  [result setValue:[NSData dataWithBytes:&data length:len]];
  return result;
 ```
 
I am not a competent ObjectiveC dev, so I am not sure if this can be optimized, perhaps there's no need at all to reallocate the above. That being said, this definitely works. The proto file for `descriptor` says the `descriptor` field should be a `bytes` or `NSData` class, not an int, so the conversion to `int` doesn't seem to make sense, at least not on the iOS version >= 12.

closed 2023-05-14 by endrelovas