pauldemarco/flutter_blue #585: Connect is never finished if device does not connect

osion-git, 2020-05-26:
Hello

**Situation:**

Bluetooth: On
Device: Try to connect
Code: `await device._bleDevice.connect(timeout: Duration(seconds: 5))`

**Problem:**

We will wait forever
`await state.firstWhere((s) => s == BluetoothDeviceState.connected);`

```
  Future<void> connect({
    Duration timeout,
    bool autoConnect = true,
  }) async {
    var request = protos.ConnectRequest.create()
      ..remoteId = id.toString()
      ..androidAutoConnect = autoConnect;

    Timer timer;
    if (timeout != null) {
      timer = Timer(timeout, () {
        disconnect();
        throw TimeoutException('Failed to connect in time.', timeout);
      });
    }

    await FlutterBlue.instance._channel
        .invokeMethod('connect', request.writeToBuffer());

    await state.firstWhere((s) => s == BluetoothDeviceState.connected);

    timer?.cancel();

    return;
  }

```

jjakob666, 2020-06-13:
This code change below solves the problem.  You need to specify the timeout on the call to state.firstWhere. Also I added a forced default timeout so it returns.  You should call the connect method with the then(), onError: pattern if you want to catch the fact that it timed out. example: 

> device.connect(timeout: Duration(seconds: 5)).then((value) async {
    //proceed to use the connetion
}, onError: (error) {
        print(error);
 });

Changed code to fix waiting forever:

> 
Future<void>` connect({
    Duration timeout = Duration(seconds:20),
    bool autoConnect = true,
  }) async {
    var request = protos.ConnectRequest.create()
      ..remoteId = id.toString()
      ..androidAutoConnect = autoConnect;

    try {
      await FlutterBlue.instance._channel
          .invokeMethod('connect', request.writeToBuffer());

      await state.firstWhere((s) => s == BluetoothDeviceState.connected).timeout(timeout);

    } catch(e) {
      print(e.toString());
      disconnect();
      return Future.error(e);
    }

    return;
  }

gujianxing, 2020-06-19:
Try use `Zone`, like this:

````
    runZoned(
      
      () async {

        Duration timeout = Duration(seconds: 5);

        await this.device.connect(
          
          timeout:timeout, 
          
          autoConnect: false

        );

      },

      onError: (error, stackTrace) {

        print(' Uncaught error: $error');
        
      }
    
    );

````

tatsuyuki25, 2020-09-08:
same problem.

Alex-Evo, 2021-04-13:
Thanks, @jjakob666, this is really an important fix. The original code throws a TimeoutException that is never caught (and somehow cannot be caught in the main application, as it appears to be limited to the plugin - not sure why that is, though). This leads to code in the main application failing due to the unhandled exception but without the chance to do something about it.

@pauldemarco: This issue is still present in the current state of the plugin, and its a pretty fundamental one. I hope it gets fixed at some point in the future, with jjakob666 having already proposed a solution.

d-wolf, 2021-04-27:
You can catch a connection timeout with the following code:

```
 runZonedGuarded(() async {
        await _current.device.connect(autoConnect: false, timeout: Duration(seconds: 5));
      }, (e, s) {
        // do sth. on timeout exception
      });
```

Alex-Evo, 2021-04-28:
Thanks, @d-wolf, that's very helpful!

BananaHemic, 2021-09-21:
I really like this repo, but I think @pauldemarco has moved on. I suggest using [this fork](https://github.com/MZachmann/flutter_blue) which includes the above change. Switching is as simple as
```
flutter_blue:
    git: https://github.com/MZachmann/flutter_blue
```
in pubspec.yaml