pauldemarco/flutter_blue #589: Error in headless/ background mode

debasmitasarkar, 2020-05-27:
flutter_blue: ^0.7.2
```
[ERROR:flutter/lib/ui/ui_dart_state.cc(166)] Unhandled Exception: MissingPluginException(No implementation found for method startScan on channel plugins.pauldemarco.com/flutter_blue/methods)
05-28 01:08:06.697  9189  9210 E flutter : #0      FlutterBlue.scan (package:flutter_blue/src/flutter_blue.dart:112:7)
05-28 01:08:06.697  9189  9210 E flutter : <asynchronous suspension>
05-28 01:08:06.697  9189  9210 E flutter : #1      FlutterBlue.startScan (package:flutter_blue/src/flutter_blue.dart:142:11)
05-28 01:08:06.697  9189  9210 E flutter : #2      bluetoothService (package:background_fetch_example/main.dart:14:15)
05-28 01:08:06.697  9189  9210 E flutter : #3      backgroundFetchHeadlessTask (package:background_fetch_example/main.dart:31:3)
05-28 01:08:06.697  9189  9210 E flutter : #4      _headlessCallbackDispatcher.<anonymous closure> (package:background_fetch/background_fetch.dart:619:15)
05-28 01:08:06.697  9189  9210 E flutter : #5      _headlessCallbackDispatcher.<anonymous closure> (package:background_fetch/background_fetch.dart:607:41)
05-28 01:08:06.697  9189  9210 E flutter : #6      MethodChannel._handleAsMethodCall (package:flutter/src/services/platform_channel.dart:409:55)
05-28 01:08:06.697  9189  9210 E flutter : #7      MethodChannel.setMethodCallHandler.<anonymous closure> (package:flutter/src/services/platform_channel.dart:377:54)
05-28 01:08:06.697  9189  9210 E flutter : #8      _DefaultBinaryMessenger.handlePlatformMessage (package:flutter/src/services/binding.dart:258:33)
05-28 01:08:06.697  9189  9210 E flutter : #9      _invoke3.<anonymous closure> (dart:ui/hooks.dart:306:15)
05-28 01:08:06.697  9189  9210 E flutter : #10     _rootRun (dart:async/zone.dart:1190:13)
05-28 01:08:06.697  9189  9210 E flutter : #11     _CustomZone.run (dart:async/zone.dart:1093:19)
05-28 01:08:06.697  9189  9210 E flutter : #12     _CustomZone.runGuarded (dart:async/zone.dart:997:7)
05-28 01:08:06.697  9189  9210 E flutter : #13     _invoke3 (dart:ui/hooks.dart:305:10)
05-28 01:08:06.697  9189  9210 E flutter : #14     _dispatchPlatformMessage (dart:ui/hooks.dart:180:5)
05-28 01:08:06.697  9189  9210 E flutter : 
05-28 01:08:11.705  9189  9189 D TSBackgroundFetch: - Background Fetch event received
05-28 01:08:11.705  9189  9189 D TSBackgroundFetch: 💀 [HeadlessTask com.transistorsoft.customtask]
05-28 01:08:11.713  9189  9210 I flutter : [BackgroundFetch] Headless event received: com.transistorsoft.customtask
05-28 01:08:11.721  9189  9210 E flutter : [ERROR:flutter/lib/ui/ui_dart_state.cc(166)] Unhandled Exception: MissingPluginException(No implementation found for method stopScan on channel plugins.pauldemarco.com/flutter_blue/methods)
05-28 01:08:11.721  9189  9210 E flutter : #0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:154:7)
05-28 01:08:11.721  9189  9210 E flutter : <asynchronous suspension>
05-28 01:08:11.721  9189  9210 E flutter : #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:329:12)
05-28 01:08:11.721  9189  9210 E flutter : #2      FlutterBlue.stopScan (package:flutter_blue/src/flutter_blue.dart:154:20)
05-28 01:08:11.721  9189  9210 E flutter : #3      bluetoothService (package:background_fetch_example/main.dart:25:15)
05-28 01:08:11.721  9189  9210 E flutter : #4      backgroundFetchHeadlessTask (package:background_fetch_example/main.dart:31:3)
05-28 01:08:11.721  9189  9210 E flutter : #5      _headlessCallbackDispatcher.<anonymous closure> (package:background_fetch/background_fetch.dart:619:15)
05-28 01:08:11.721  9189  9210 E flutter : #6      _headlessCallbackDispatcher.<anonymous closure> (package:background_fetch/background_fetch.dart:607:41)
05-28 01:08:11.721  9189  9210 E flutter : #7      MethodChannel._handleAsMethodCall (package:flutter/src/services/platform_channel.dart:409:55)
05-28 01:08:11.721  9189  9210 E flutter : #8      MethodChannel.setMethodCallHandler.<anonymous closure> (package:flutter/src/services/platform_channel.dart:377:54)
05-28 01:08:11.721  9189  9210 E flutter : #9      _DefaultBinaryMessenger.handlePlatformMessage (package:flutter/src/services/binding.dart:258:33)
05-28 01:08:11.721  9189  9210 E flutter : #10     _invoke3.<anonymous closure> (dart:ui/hooks.dart:306:15)
05-28 01:08:11.721  9189  9210 E flutter : #11     _rootRun (dart:async/zone.dart:1190:13)
05-28 01:08:11.721  9189  9210 E flutter : #12     _CustomZone.run (dart:async/zone.dart:1093:19)
05-28 01:08:11.721  9189  9210 E flutter : #13     _CustomZone.runGuarded (dart:async/zone.dart:997:7)
05-28 01:08:11.721  9189  9210 E flutter : #14     _invoke3 (dart:ui/hooks.dart:305:10)
05-28 01:08:11.721  9189  9210 E flutter : #15     _dispatchPlatformMessage (dart:ui/hooks.dart:180:5)
05-28 01:08:11.721  9189  9210 E flutter : 
05-28 01:08:11.721  9189  9210 I flutter : Error starting scan.
```

debasmitasarkar, 2020-05-27:
@pauldemarco  please help fixing the issue. Thanks in advance

Errichamonda, 2020-07-15:
Any news on this?

DanielPiflaks, 2021-08-17:
@debasmitasarkar I'm facing the same issue...
Did you find any solution?