pauldemarco/flutter_blue #590: 我该如何获取自己的设备蓝牙mac地址？

812781385, 2020-05-28:
我如何拿到自己的蓝牙mac地址

itMcdull, 2020-05-28:
device.id

812781385, 2020-05-29:
> 设备ID

具体怎么写法啊？只有这样的一个实例：flutterBlue = FlutterBlue.instance;

812781385, 2020-05-29:
> device.id

具体怎么写法啊？只有这样的一个实例：flutterBlue = FlutterBlue.instance;

itMcdull, 2020-05-29:
看文档。先扫描啊，然后拿到scanResults的返回值，安卓才能拿到mac。ios拿不到，你可以让你们后台拿到mac地址在name返回给你，你自己拼接。

812781385, 2020-05-29:
> 看文档。先扫描啊，然后拿到scanResults的返回值，安卓才能拿到mac。ios拿不到，你可以让你们后台拿到mac地址在name返回给你，你自己拼接。

扫描出来的是别人的设备，我是需要本机的蓝牙地址哦。

itMcdull, 2020-05-29:
> > device.id
> 
> 具体怎么写法啊？只有这样的一个实例：flutterBlue = FlutterBlue.instance;

Suggest  issue to post in English

itMcdull, 2020-05-29:
> > 看文档。先扫描啊，然后拿到scanResults的返回值，安卓才能拿到mac。ios拿不到，你可以让你们后台拿到mac地址在name返回给你，你自己拼接。
> 
> 扫描出来的是别人的设备，我是需要本机的蓝牙地址哦。

device_info去看下没有吧，你要获取自己的蓝牙mac地址有啥用

LongXiangGuo, 2021-05-17:
iOS 拿不到，corebluetooth 的api都没有暴露出来。

Android 默认的api可以取到本机蓝牙设备的  address 和 name

```
     BluetoothAdapter bluetoothAdapter = BluetoothAdapter.getDefaultAdapter();
        @RequiresPermission(Manifest.permission.BLUETOOTH)
        public String getAddress() {
            try {
                return mManagerService.getAddress();
            } catch (RemoteException e) {
                Log.e(TAG, "", e);
            }
            return null;
        }

        /**
         * Get the friendly Bluetooth name of the local Bluetooth adapter.
         * <p>This name is visible to remote Bluetooth devices.
         * <p>Requires {@link android.Manifest.permission#BLUETOOTH}
         *
         * @return the Bluetooth name, or null on error
         */
        public String getName() {
            try {
                return mManagerService.getName();
            } catch (RemoteException e) {
                Log.e(TAG, "", e);
            }
            return null;
        }
        
```