pauldemarco/flutter_blue #591: Why can I get "scanresult.rssi" and "device.id", but I can't get "device.name"?

XiaoQihang, 2020-05-29:
Has anyone ever seen this happen?Thank you very much
`    var scanSubscription = flutterBlue.scan().listen((scanResult) {
      // do something with scan result
      var device;
      device = scanResult.device;
      print( '${device.name} found! rssi: ${scanResult.rssi},address:${device.id}');
    });`
got
`I/flutter (20841):  found! rssi: -67,address:1D:C0:CF:76:A4:DE
I/flutter (20841):  found! rssi: -61,address:25:2E:24:0A:68:A8
I/flutter (20841):  found! rssi: -90,address:00:3C:D2:52:16:2E`

XiaoQihang, 2020-06-01:
The previous version of flutter was 1.12.13+hotfix8 flutter_blue version 6.2.0, and the update of flutter to 1.12.13+hotfix9 and flutter_blue7.0.2 was normal

LinJinghuidev, 2020-07-23:
It doesn't work

florianlenz, 2020-09-02:
Having the same problem on my google pixel.

Rayen-cherni, 2020-09-04:
I have the same problem.
who fixed it ?