pauldemarco/flutter_blue #592: E/flutter (11239): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: PlatformException(reconnect_error, error when reconnecting to device, null)

itMcdull, 2020-05-29:
Manually disconnect the device and reconnect error
```
E/BluetoothGatt(11239):
E/BluetoothGatt(11239): android.os.DeadObjectException
E/BluetoothGatt(11239): 	at android.os.BinderProxy.transactNative(Native Method)
E/BluetoothGatt(11239): 	at android.os.BinderProxy.transact(BinderProxy.java:526)
E/BluetoothGatt(11239): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.clientConnect(IBluetoothGatt.java:2003)
E/BluetoothGatt(11239): 	at android.bluetooth.BluetoothGatt.connect(BluetoothGatt.java:929)
E/BluetoothGatt(11239): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.onMethodCall(FlutterBluePlugin.java:300)
E/BluetoothGatt(11239): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:231)
E/BluetoothGatt(11239): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:93)
E/BluetoothGatt(11239): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:642)
E/BluetoothGatt(11239): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/BluetoothGatt(11239): 	at android.os.MessageQueue.next(MessageQueue.java:363)
E/BluetoothGatt(11239): 	at android.os.Looper.loop(Looper.java:173)
E/BluetoothGatt(11239): 	at android.app.ActivityThread.main(ActivityThread.java:8147)
E/BluetoothGatt(11239): 	at java.lang.reflect.Method.invoke(Native Method)
E/BluetoothGatt(11239): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
E/BluetoothGatt(11239): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1101)
E/flutter (11239): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: PlatformException(reconnect_error, error when reconnecting to device, null)
E/flutter (11239): #0      StandardMethodCodec.decodeEnvelope 
package:flutter/…/services/message_codecs.dart:569
E/flutter (11239): #1      MethodChannel.invokeMethod 
package:flutter/…/services/platform_channel.dart:321
E/flutter (11239): <asynchronous suspension>
E/flutter (11239): #2      BluetoothDevice.connect 
package:flutter_blue/src/bluetooth_device.dart:38
E/flutter (11239): #3      BLueDevice.connect 
package:gl_inet/…/providers/blue_device.dart:18
E/flutter (11239): #4      _BlueInitializationState._getBlueIsScaning.<anonymous closure>.<anonymous closure>.<anonymous closure>.<anonymous closure> 
package:gl_inet/…/Initialization/blue_Initialization.dart:162
E/flutter (11239): #5      GestureRecognizer.invokeCallback 
package:flutter/…/gestures/recognizer.dart:182
E/flutter (11239): #6      TapGestureRecognizer.handleTapUp 
package:flutter/…/gestures/tap.dart:486
E/flutter (11239): #7      BaseTapGestureRecognizer._checkUp 
package:flutter/…/gestures/tap.dart:264
E/flutter (11239): #8      BaseTapGestureRecognizer.acceptGesture 
package:flutter/…/gestures/tap.dart:236
E/flutter (11239): #9      GestureArenaManager.sweep 
package:flutter/…/gestures/arena.dart:156
E/flutter (11239): #10     GestureBinding.handleEvent 
package:flutter/…/gestures/binding.dart:222
E/flutter (11239): #11     GestureBinding.dispatchEvent 
package:flutter/…/gestures/binding.dart:198
E/flutter (11239): #12     GestureBinding._handlePointerEvent 
package:flutter/…/gestures/binding.dart:156
E/flutter (11239): #13     GestureBinding._flushPointerEventQueue 
package:flutter/…/gestures/binding.dart:102
E/flutter (11239): #14     GestureBinding._handlePointerDataPacket 
package:flutter/…/gestures/binding.dart:86
E/flutter (11239): #15     _rootRunUnary  (dart:async/zone.dart:1138:13)
E/flutter (11239): #16     _CustomZone.runUnary  (dart:async/zone.dart:1031:19)
E/flutter (11239): #17     _CustomZone.runUnaryGuarded  (dart:async/zone.dart:933:7)
E/flutter (11239): #18     _invoke1  (dart:ui/hooks.dart:273:10)
E/flutter (11239): #19     _dispatchPointerDataPacket  (dart:ui/hooks.dart:182:5)
E/flutter (11239):

```
What problem and how to solve it
thx

itMcdull, 2020-05-29:
```
Flutter 1.12.13+hotfix.9 • channel unknown • unknown source
Framework • revision f139b11009 (8 weeks ago) • 2020-03-30 13:57:30 -0700
Engine • revision af51afceb8
Tools • Dart 2.7.2
```
```
flutter_blue: 0.7.2
```

btbj, 2020-05-29:
```
Flutter (Channel unknown, v1.10.14, on Mac OS X 10.15.4 19E287, locale zh-Hans-CN)
    • Flutter version 1.10.14
    • Framework revision 1946fc4da0 (8 months ago), 2019-10-07 15:23:31 -0700
    • Engine revision 1d62160fdb
    • Dart version 2.6.0 (build 2.6.0-dev.1.0 d6c6d12ebf)
Xcode - develop for iOS and macOS (Xcode 11.4.1)
Android Studio (version 3.4)
```

```
flutter_blue: ^0.6.3+1
```

also met PlatformException(reconnect_error, error when reconnecting to device, null)

itMcdull, 2020-05-29:
> ```
> Flutter (Channel unknown, v1.10.14, on Mac OS X 10.15.4 19E287, locale zh-Hans-CN)
>     • Flutter version 1.10.14
>     • Framework revision 1946fc4da0 (8 months ago), 2019-10-07 15:23:31 -0700
>     • Engine revision 1d62160fdb
>     • Dart version 2.6.0 (build 2.6.0-dev.1.0 d6c6d12ebf)
> Xcode - develop for iOS and macOS (Xcode 11.4.1)
> Android Studio (version 3.4)
> ```
> 
> ```
> flutter_blue: ^0.6.3+1
> ```
> 
> 还遇到PlatformException（reconnect_error，重新连接到设备时出错，null）

+1

bsmerbeck, 2020-07-27:
I've also encountered this bug using 0.7.2.

Running on Pixel 3a running android 10.

In my case, this exception is called when bluetooth is disabled and then reenabled.

jimykhan, 2021-01-28:
> I've also encountered this bug using 0.7.2.
> 
> Running on Pixel 3a running android 10.
> 
> In my case, this exception is called when bluetooth is disabled and then reenabled.
> 
> I'm also storing devices using `shared-preferences` so that they can be reconnected to or other attempted to be connected.

same issue with please can u guide me if solve this issue

JKornberg, 2021-06-25:
Still have this issue on Android only

dudizimber, 2021-08-10:
Do we have a solution for that?

MarkMacDon, 2021-12-17:
> 

I am having the same issue. Please let me know if you find a solution

shuizhuyu1015, 2021-12-26:
I am having the same issue, too. Did you solve it?