pauldemarco/flutter_blue #593: 13.4.1

wolfc01, 2020-05-29:
Hi,

I made an BT app for iOS / Android 

The app runs fine on iOS version 11.2.5;
The app runs fine on Android;
BUT: The app does run also on Iphone XR, ios 13.4.1, but does not detect any BLE device.

I updated the Info.plist according to your repo, which fixed a crash on ios 13.4.1, but still no BLE devices detected.

Any hint?

Best regards
Carl.

robinjohn, 2020-08-16:
knock.. knock.. 
Any progress you had @wolfc01 ? 
I face similar issue with a BLE device not detected on iOS (13.5.1 on IPhone7), but discovering on Android (Android v9 on S8).

wolfc01, 2020-08-16:
Hi,

No progress yet as of lack of time and having an apple device...

Br
Carl