pauldemarco/flutter_blue #594: can not build release build on iOS

vaimikpatel2908, 2020-05-29:
[image]

I have already added this files into debugIphoneos,releaseIphoneos folders when I was building the debug and release builds respectively but now I was trying to build final release build but now I don't know where this file is missing. Using Flutter's latest stable build 1.17.2.

gmaxferr, 2020-06-16:
i had the same issue, using the Legacy build system and cocoapods v1.7.5 solved it for me

netskink, 2020-06-26:
> i had the same issue, using the Legacy build system and cocoapods v1.7.5 solved it for me

Can you elaborate on this method please?