pauldemarco/flutter_blue #595: do not work disconnect

minoto9, 2020-05-30:
I join to app this plugin, but not good work disconnect. please help.
first once connecting, and requesting is OK.
but then disconnect and continue connect is not working.
 
```
    try {
      // scan device
      BleStandPeripheral device = await _scan(standID);
      // when I debug, second connect stop there
      this.central.connect(device);
      completer.complete("connected");
    } catch (e) {
      logger.warning(e.toString());
      completer.completeError(e.toString());
    }
```

```
Future<BleStandPeripheral> _scan(String standID) {
    logger.info("scanning... $standID");
    var completer = Completer<BleStandPeripheral>();

    _observePeripheral = this.central.scannedDevice;
    _observePeripheral
        .where((device) => standID == device.name)
        .listen((device) {
      this.central.stopScan();
      logger.info("scanned device: ${device.name}");
      completer.complete(device);
    }).onError((err) {
      logger.info(err.toString());
      completer.completeError(err.toString());
    });

    this.central.startDeviceScan();
    return completer.future;
  }
```

philos3, 2020-06-23:
same to me , the problem is disconnect did not work well everytime(are you disconnect in dispose or deactivate,sometimes it did not execute in time ) ,so continue connect is not working somtimes

,

liraku, 2020-08-11:
I ran into the same problem. Ask for help