pauldemarco/flutter_blue #597: The eigenvalue indicates that each indication will carry the data of the last time

Merely-chao, 2020-06-01:


Merely-chao, 2020-06-01:
OS : IOS 13.5
`flutter: BluetoothCharacteristic{uuid: 00002a35-0000-1000-8000-00805f9b34fb, deviceId: 8E5A8283-FF3F-0360-8CEA-9C9D2267E04A, serviceUuid: 00001810-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: false, notify: false, indicate: true, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, descriptors: [BluetoothDescriptor{uuid: 00002902-0000-1000-8000-00805f9b34fb, deviceId: 8E5A8283-FF3F-0360-8CEA-9C9D2267E04A, serviceUuid: 00001810-0000-1000-8000-00805f9b34fb, characteristicUuid: 00002a35-0000-1000-8000-00805f9b34fb, value: [0, 0, 0, 0]}], value: [22, 102, 0, 60, 0, 72, 0, 228, 7, 6, 1, 17, 51, 29, 63, 0, 0, 0]
flutter: BluetoothCharacteristic{uuid: 00002a35-0000-1000-8000-00805f9b34fb, deviceId: 8E5A8283-FF3F-0360-8CEA-9C9D2267E04A, serviceUuid: 00001810-0000-1000-8000-00805f9b34fb, secondaryServiceUuid: null, properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: false, notify: false, indicate: true, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, descriptors: [BluetoothDescriptor{uuid: 00002902-0000-1000-8000-00805f9b34fb, deviceId: 8E5A8283-FF3F-0360-8CEA-9C9D2267E04A, serviceUuid: 00001810-0000-1000-8000-00805f9b34fb, characteristicUuid: 00002a35-0000-1000-8000-00805f9b34fb, value: [2, 0, 0, 0]}], value: [22, 102, 0, 60, 0, 72, 0, 228, 7, 6, 1, 17, 51, 29, 63, 0, 0, 0]
flutter: indicate:[22, 102, 0, 60, 0, 72, 0, 228, 7, 6, 1, 17, 51, 29, 63, 0, 0, 0]
flutter: indicate:[22, 108, 0, 67, 0, 87, 0, 228, 7, 6, 1, 17, 53, 5, 64, 0, 0, 0]
`

danasorani, 2020-07-28:
i have same issue , did you fix it ?