pauldemarco/flutter_blue #600: BluetoothDeviceState StreamSubscription - Fires duplicate event for every resubscription.

fyredave, 2020-06-03:
On a Google Pixel 2 XL running Android 10

When reconnecting to the same device and listening to the BluetoothDeviceState

`StreamSubscription<BluetoothDeviceState> _stateSubscription;`

When I disconnect I call
`await _stateSubscription?.cancel(); _stateSubscription = null;`

When I reconnect a few seconds later I call
`_stateSubscription = _device.state.listen(_handleBLEStateChanged);`

after that `_handleBLEStateChanged `is fired twice. If I disconnect and reconnect it fires a third time.

It continues to fire an additional duplicate event every time I reconnect

I have tried using single stream subscription with onPause and onResume but it has the same effect. It's possible it's the same error in the stream logic that was discovered here: https://github.com/pauldemarco/flutter_blue/issues/525

a564068328, 2020-07-03:
This situation sometimes leads to a situation where the ble device is connected but the status is incorrect

bsmerbeck, 2020-07-28:
+1

I have a device that fails to connect on the first attempt. As such, it always ends up with multiple firings that grow with each connection.

flutter_blue 0.7.2, flutter 1.17.15, running on a Pixel 3A android 10

gksrb4, 2020-09-09:
+1

how can you solve this?

gksrb4, 2020-09-09:
I find [https://github.com/pauldemarco/flutter_blue/issues/317#issuecomment-607268143](url)

claudioafelix, 2020-12-08:
I have the same situations, and I don't know if it's an error or just bad usage!

I am running:
- fllutter_blue 0.7.3
- Flutter 1.22.4 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 1aafb3a8b9 (4 weeks ago) • 2020-11-13 09:59:28 -0800
Engine • revision 2c956a31c0
Tools • Dart 2.10.4
- The device is a Samsung A10 with android 10.

I tried the solution [https://github.com/pauldemarco/flutter_blue/issues/317#issuecomment-607268143](url)  but didn't work.

fyredave, 2021-01-05:
I've submitted a pull request that seems to have solved our issues with this, I haven't been able to reproduce this issue since.