pauldemarco/flutter_blue #611: Json value gets cut off

mbostic, 2020-06-17:
hi, I have asked a question on Stack Exchange, but I think I should also post the problem here.
The link is: [](https://stackoverflow.com/questions/62436798/flutter-blue-json-value-gets-cut-off)

nmbus, 2020-06-18:
Hi, the problem definitely comes from the ATT MTU that is likely negociated at its minimum (23 bytes), which with the ATT header for a Notification (3 bytes) makes you able to only exchange 20 bytes at once, hence why your notification gets truncated at 20 bytes. Either find a way to negociate a higher MTU (can usually go up to 512 bytes), or find a way to trigger a characteristic long read, ie the central should issue read (blob) requests after a notification with a filled PDU is received. Not sure if iOS/Android support such behavior.

mbostic, 2020-06-18:
@nmbus Thanks, I had to use `await device.requestMtu(512);` and now it works.

closed 2020-06-18 by mbostic

arizki29, 2020-07-10:
i have same isue.  
where should i put the code (await device.requestMtu(512);)  ??? 
thx

mbostic, 2020-07-10:
@arizki29 You should put it after you connect to the device ( after `await device.connect();`)

arizki29, 2020-07-13:
@mbostic  I have tried to put it here after "await device.connect();"  but the results always return to default 23.
this my result:

D/FlutterBluePlugin(27476): [onServicesDiscovered] count: 3 status: 0
D/BluetoothGatt(27476): configureMTU() - device: 4C:24:98:77:6F:B1 **mtu: 512**
D/BluetoothGatt(27476): setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: true
I/flutter (27476):
D/BluetoothGatt(27476): onConfigureMTU() - Device=4C:24:98:77:6F:B1 **mtu=23 status=0**
D/FlutterBluePlugin(27476): [onMtuChanged] **mtu: 23 status: 0**

note : flutter_blue: ^0.7.2

yasasmi, 2021-04-18:
@nmbus   hi I'm using flutter_blue forget data from my MIband 5 but it gave me an empty array and disconnecting the device how can I solve this issue cus I cant understand what is the specific root for this matter. if I can get help it will be really helpfi. thank you!

Logamuhilan, 2021-12-10:
hi there. I am new to flutter : ) I am using flutter_blue to connect and get bluetooth device data, my problem is i dont know the method to get data from it. please give me some methods to get it. i have used "await c.read" it gives only device name, and "await d.read" it gives only empty array. here "c" and "d" is characteristics. thanks in advance ; )