pauldemarco/flutter_blue #617: run failures with iOS

netskink, 2020-06-26:
Using the example as a guide

0. specify channel and update

I tried stable.  It failed.  Here I am trying dev channel.

```
flutter channel dev
flutter upgrade
```

1.  Create a stub project 
```
flutter create bt_testy
cd bt_testy
open ios/Runner.xcworkspace/
# correct bundle identifier  and signing team name
# run sample on phone via Xcode. close and test with cli
# since there are errors later with ios8.0 vs iOS 10.0 capabilities, arbitrarily change to iOS 13
flutter run
```
2.  Modify pubspec.yaml file so the dependency section looks like this:
```
dependencies:
  flutter_blue: ^0.7.2
  flutter:
    sdk: flutter
```

3.  Modify the lib/main.dart file

```
# Replace the contents of lib/main.dart with demo code and replace with example code 
# https://pub.dev/packages/flutter_blue#-example-tab-
```

4. Clean and install dependencies

```
flutter clean
flutter pub get
flutter clean
```

5.  Attempt to build and run the example code

```
flutter run
```

I read online about removing my test/ dir and lib/main.dart with the two files listed in the tar file.    Lastly I kept the widgets.dart file form the example and used the original main.dart file.

The result is that the code builds and starts but quickly crashes.  If I run the app in Xcode, it crashes and gives a crash dump.

I'm new to flutter so perhaps this is an easily resolved error.

davis-c, 2020-07-01:
I created stackoverflow and added a bounty for this issue.  See here https://stackoverflow.com/questions/62638588/how-to-use-flutter-blue-with-ios

davis-c, 2020-07-04:
@blaugold helped me with this.  I updated the stackoverflow post.  Here is the  solution from SO.

I tried a friends version of the code that worked for him but not me.

See [here][1]

Since that code failed in the same spot, I examined the stack trace some more and I noticed an earlier failure in the stack trace.

```
TCC`__CRASHING_DUE_TO_PRIVACY_VIOLATION__:
```

As a result, I added the following 

```
<key>NSBluetoothAlwaysUsageDescription</key>
    <string>This app uses bluetooth</string>
```

To the `ios/Runner/Info.plist file`. 

After that it works.

  [1]: https://github.com/blaugold

closed 2020-07-04 by netskink