pauldemarco/flutter_blue #622: /Users/flutter/flutter/.pub-cache/hosted/pub.flutter-io.cn/flutter_blue-0.7.2/ios/gen/Flutterblue.pbobjc.h: No such file or directory (in target 'flutter_blue' from project 'Pods')

GeminiGuoGuo, 2020-06-30:
i have a problem when i run on xcode,how to solve.

maqoo, 2020-07-08:
I was able to work around this by removing the file from build phases of flutter_blue.framework in Pods project. After that I just copied and added it to Pods project and removed "gen/" prefix in import.

Then only I was able to run project from xcode. This wasn't needed to run from Android Studio as the pods are just installed and Pods project is not built when running app.

dspells, 2020-10-22:
I'm having the same problem.

dspells, 2021-06-24:
`open /.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.8.0/ios/gen/Flutterblue.pbobjc.h: No such file or directory`

The project keeps getting into this state and I'm not sure why.
I'm using Xcode 12.5.1 now flutter 2.2.2 dart 2.13.3