pauldemarco/flutter_blue #625: Wait for scan to complete

CLDiego, 2020-07-02:
Hi, I'm exploring using flutter_blue for my application. My idea is to have a scan button and let the user choose to which device to connect after. Is there a way to know when the scan has been completed? When calling the scan method I set the duration to 3 seconds. I wanted to know if there is a dynamic way to resolve this issue or if it's a matter of hardcoding a 3 second delay.

So far I have tried:

`    await for (List result in ble.scanResults){
      print(result);
    }`

but the code does not stop and wait for the scan to be completed.

I've also tried looking at the isScanning flag:
`    ble.isScanning.listen((event) {
      if(!event){
        print('scan complete');
      }
    });`

But this approach does not work since it seems that even when the startScan() method is called, the flag takes some time to change. Any suggestions are welcomed!

StephanieGitHub, 2020-07-04:
I have the same question, How to know when the scan is completed?

Zcating, 2020-07-06:
Just use `startScan` method like following:
```
await startScan(timeout: const Duration(seconds: 3));
```

Then, use `scanResults` to observe  the stream for devices.

CLDiego, 2020-07-07:
HI @Zcating, I just tested your solution. It works nicely. 

I was confused on where to `await` initially I tried directly on the scanResults but your solution makes more sense.

Thanks!

closed 2020-07-07 by CLDiego