pauldemarco/flutter_blue #627: Is it possible to connect to some device without scan? I have the MAC ADDRESS

Luiggi13, 2020-07-07:
I have an IoT Lock and I want to connect it witouth scan. I have the Mac Address.
I need to achieve it without scan.

The steps that I'm trying to do are:

-Starting the app
-Connect to my device
-Retrieve the lock status (opened or closed).

Thanks to all

boskokg, 2020-07-07:
Yes, but only on android

Luiggi13, 2020-07-08:
> Yes, but only on android

Thanks, I'm trying to it, a week ago... 

Can you post some code of example

hahazheng, 2020-07-14:
@Luiggi13 
 I have the same problem as you. Is there a good solution for locks?

Luiggi13, 2020-07-14:
Yes , it's good, works well, but I need to connect to device by id. I don't want to scan. 'Cause could have some property with N locks. 

You have been able to connect directly to only one device?

hahazheng, 2020-07-14:
@Luiggi13 I can connect to the device, I have multiple devices, but I have a problem when unlocking. Can i get a contact for you。Thanks

Luiggi13, 2020-07-14:
Yes no problem.
To do something with the lock, you must send bytes like this: 

`await c.write([0x00, 0xAA, 0x54, 0x45, 0x53, 0x54],
                          withoutResponse: true)`

gmail: rnative.cllansola@gmail.com

Luiggi13, 2020-07-27:
> Yes, but only on android

@boskokg can you write me some basic code for connect to a specific device.id???
I would be appreciated

Thanks a lot

boskokg, 2020-07-27:
@Luiggi13 this is part of the code for that purpose:

    if (Platform.isAndroid) {
      protos.BluetoothDevice dev = protos.BluetoothDevice()..remoteId = address;
      _gattClient = BluetoothDevice.fromProto(dev);

      if (!_stateListeners.containsKey(_gattClient)) {
        _stateListeners[_gattClient] = _gattClient.state
            .skip(1)
            .listen((BluetoothDeviceState newState) => _onConnectionStateChange(_gattClient, newState));
      }
      _gattClient.connect(autoConnect: BluetoothHelper.everConnectedDevicesAddresses.contains(address));
    }

boskokg, 2020-07-27:
Basically, you need this:

  protos.BluetoothDevice dev = protos.BluetoothDevice()..remoteId = address;
  _gattClient = BluetoothDevice.fromProto(dev);

Luiggi13, 2020-07-27:
> Basically, you need this:
> 
> protos.BluetoothDevice dev = protos.BluetoothDevice()..remoteId = address;
> _gattClient = BluetoothDevice.fromProto(dev);

Interesting... thanks, I'll try it

Luiggi13, 2020-07-27:
> protos.BluetoothDevice dev = protos.BluetoothDevice()..remoteId = address;
> _gattClient = BluetoothDevice.fromProto(ddev);

How to fix it? It's my first time with it! 👍 
[image]

boskokg, 2020-07-28:
import 'package:flutter_blue/gen/flutterblue.pb.dart' as protos;

Luiggi13, 2020-07-28:
> import 'package:flutter_blue/gen/flutterblue.pb.dart' as protos;

@boskokg thanks a lot, I'll try it. Sorry for the messages!

Luiggi13, 2020-07-30:
It's working thanks, man!!!!!!!!

Luiggi13, 2020-07-30:
Hi people, the code works properly but now I'm trying to compile apk, and I have this issue.
I've been check this issue too: https://github.com/pauldemarco/flutter_blue/issues/141

`Running Gradle task 'assembleRelease'...                                
Running Gradle task 'assembleRelease'... Done                      70,8s
The built failed likely due to AndroidX incompatibilities in a plugin. The tool is about to try using Jetfier to solve the incompatibility.
Building plugin flutter_blue...
Running Gradle task 'assembleAarRelease'...                             
Running Gradle task 'assembleAarRelease'... Done                    1,4s

> Configure project :
The com.google.protobuf plugin was already applied to the project: : and will not be applied again after plugin: android-library
         *********************************************************
WARNING: This version of flutter_blue will break your Android build if it or its dependencies aren't compatible with AndroidX.
         See https://goo.gl/CP92wY for more information on the problem and how to fix it.
         This warning prints for all Android build failures. The real root cause of the error may be unrelated.
         *********************************************************

FAILURE: Build failed with an exception.

* What went wrong:
Task 'assembleAarRelease' not found in root project 'flutter_blue'.

* Try:
Run gradlew tasks to get a list of available tasks. Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s

The plugin flutter_blue could not be built due to the issue above.`

Somebody know how can fix it?

Luiggi13, 2020-07-30:
More info

**Note: /Users/bebito/development/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.2/android/src/main/java/com/pauldemarco/flutter_blue/FlutterBluePlugin.java uses or overrides a deprecated API.**
Note: Recompile with -Xlint:deprecation for details.                    
Note: /Users/bebito/Downloads/demoflutter/kiploc/build/flutter_blue/generated/source/proto/release/javalite/com/pauldemarco/flutter_blue/Protos.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.                      
Removed unused resources: Binary resource data reduced from 483KB to 474KB: Removed 1%
                                                                        
FAILURE: Build failed with an exception.                                
                                                                        
* What went wrong:                                                      
Execution failed for task ':system_setting:verifyReleaseResources'.     
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > Android resource linking failed                                    
     /Users/bebito/.gradle/caches/transforms-2/files-2.1/a528b13ac93e64cafa3d0480e2c93207/core-1.1.0/res/values/values.xml:142:5-173:25: AAPT: error: resource android:attr/fontVariationSettings not found.
                                                                        
     /Users/bebito/.gradle/caches/transforms-2/files-2.1/a528b13ac93e64cafa3d0480e2c93207/core-1.1.0/res/values/values.xml:142:5-173:25: AAPT: error: resource android:attr/ttcIndex not found.
                                                                        
                                                                        
* Try:                                                                  
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
                                                                        
* Get more help at https://help.gradle.org                              
                                                                        
BUILD FAILED in 1m 11s                                                  
Running Gradle task 'assembleRelease'...                                
Running Gradle task 'assembleRelease'... Done                      77,0s
The built failed likely due to AndroidX incompatibilities in a plugin. The tool is about to try using Jetfier to solve the incompatibility.
Building plugin flutter_blue...
Running Gradle task 'assembleAarRelease'...                             
Running Gradle task 'assembleAarRelease'... Done                    1,6s
         *********************************************************
WARNING: This version of flutter_blue will break your Android build if it or its dependencies aren't compatible with AndroidX.
         See https://goo.gl/CP92wY for more information on the problem and how to fix it.
         This warning prints for all Android build failures. The real root cause of the error may be unrelated.
         *********************************************************

FAILURE: Build failed with an exception.

* What went wrong:
Task 'assembleAarRelease' not found in root project 'flutter_blue'.

* Try:
Run gradlew tasks to get a list of available tasks. Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s

The plugin flutter_blue could not be built due to the issue above.

Luiggi13, 2020-08-01:
Just must to update minSdkVersion and targetSdkVersion like this and works:

        minSdkVersion 19
        targetSdkVersion 29

staminna, 2021-09-16:
This should no longer work with version 30.
Cannot roll back since the the latest version on the repository uses 30 in 6 different places and will break with 29.

Please help