pauldemarco/flutter_blue #629: Not Building iOS In Flutter Module Project

aknolldev, 2020-07-09:
I want to use this package in a Flutter module that will be embedded via AAR (Android) and XCFramework (iOS) into native iOS and Android host Apps (Swift / Kotlin). For Android, it seems to be working well but I am not able to build the iOS App (neither directly nor as a XCFramework). Using the package in a full Flutter App works. Can you please tell me what I am doing wrong?

**Flutter Doctor**
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, v1.17.5, on Mac OS X 10.14.6 18G3020, locale de-DE)
 
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.3)
[✓] Xcode - develop for iOS and macOS (Xcode 11.3.1)
[✓] Android Studio (version 3.6)
[✓] Connected device (1 available)

• No issues found!

**When building the module app for iOS Simulator using Android studio**
Launching lib/main.dart on iPhone 11 Pro in debug mode...
Running Xcode build...
Xcode build done.                                            2,3s
Failed to build iOS app
Error output from Xcode build:
↳
    ** BUILD FAILED **

Xcode's output:
↳
    While building module 'flutter_blue' imported from /Users/<user>/Android Studio Projects/flutter_module_blue_test/.ios/Flutter/FlutterPluginRegistrant/Classes/GeneratedPluginRegistrant.m:8:
    While building module 'protobuf' imported from /Users/<user>/Android Studio Projects/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.2/ios/gen/Flutterblue.pbobjc.h:13:
    In file included from <module-includes>:1:
    In file included from /Users/<user>/Android Studio Projects/flutter_module_blue_test/.ios/Pods/Target Support Files/Protobuf/Protobuf-umbrella.h:28:
    In file included from /Users/<user>/Android Studio Projects/flutter_module_blue_test/.ios/Pods/Protobuf/objectivec/GPBProtocolBuffers.h:44:
    /Users/<user>/Android Studio Projects/flutter_module_blue_test/.ios/Pods/Protobuf/objectivec/GPBWellKnownTypes.h:44:10: fatal error: 'google/protobuf/Any.pbobjc.h' file not found
     #import "google/protobuf/Any.pbobjc.h"
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    While building module 'flutter_blue' imported from /Users/<user>/Android Studio Projects/flutter_module_blue_test/.ios/Flutter/FlutterPluginRegistrant/Classes/GeneratedPluginRegistrant.m:8:
    In file included from <module-includes>:1:
    In file included from /Users/<user>/Android Studio Projects/flutter_module_blue_test/.ios/Pods/Target Support Files/flutter_blue/flutter_blue-umbrella.h:14:
    /Users/<user>/Android Studio Projects/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.2/ios/gen/Flutterblue.pbobjc.h:13:10: fatal error: could not build module 'protobuf'
     #import "GPBProtocolBuffers.h"
      ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
    2 errors generated.
    /Users/<user>/Android Studio Projects/flutter_module_blue_test/.ios/Flutter/FlutterPluginRegistrant/Classes/GeneratedPluginRegistrant.m:8:9: fatal error: could not build module 'flutter_blue'
    #import <flutter_blue/FlutterBluePlugin.h>
     ~~~~~~~^
    3 errors generated.
    note: Using new build system
    note: Planning build
    note: Constructing build description

Could not build the application for the simulator.
Error launching application on iPhone 11 Pro.

**When building the module as a XCFramework**
 flutter build ios-framework --xcframework --no-universal
Building frameworks for com.test.smarthome.v2.fluttermodulebluetest in debug mode...
 ├─Populating Flutter.framework...                                  0,3s
 ├─Creating Flutter.xcframework...                                  1,5s
 ├─Adding placeholder App.framework for debug...                   172ms
 ├─Assembling Flutter resources for App.framework...                7,7s
 ├─Creating App.xcframework...                                      0,9s
 ├─Building plugins...                                              7,3s
Unable to build plugin frameworks: ** BUILD FAILED **

The following build commands failed:
        CompileC /Users/<user>/Android\ Studio\
        Projects/flutter_module_blue_test/build/ios/framework/Debug/iphoneos/Pods.build/Debug-iphoneos/FlutterPluginRegistrant.build/Objects-normal/armv7/GeneratedPluginRegistrant.o
        /Users/<user>/Android\ Studio\ Projects/flutter_module_blue_test/.ios/Flutter/FlutterPluginRegistrant/Classes/GeneratedPluginRegistrant.m normal armv7 objective-c
        com.apple.compilers.llvm.clang.1_0.compiler
        CompileC /Users/<user>/Android\ Studio\
        Projects/flutter_module_blue_test/build/ios/framework/Debug/iphoneos/Pods.build/Debug-iphoneos/FlutterPluginRegistrant.build/Objects-normal/arm64/GeneratedPluginRegistrant.o
        /Users/<user>/Android\ Studio\ Projects/flutter_module_blue_test/.ios/Flutter/FlutterPluginRegistrant/Classes/GeneratedPluginRegistrant.m normal arm64 objective-c
        com.apple.compilers.llvm.clang.1_0.compiler
(2 failures)

Ahmadre, 2020-07-11:
same problem here. Can't use it for ios...

chaochaox1990, 2020-07-15:
add +1

aBuder, 2020-07-16:
Same problem

aknolldev, 2020-07-18:
I actually found a workaround that I want to share with you:

1) Try to run the app in an iOS simulator from Android Studio -> errors occur
2) Right click on the ".iOS" folder -> Flutter -> Open in XCode
3) Try to run the app in an iOS simulator from XCode -> errors occur
4) Click on the first error (google/protobuf/Any.pbobjc.h file not found) -> a file opens
5) Delete "google/protobuf/" in the lines 44,45 and 46 of the opened file
6) Try to run the app again in XCode -> errors occur
7) Click on the first error  (google/protobuf/Any.pbobjc.h file not found) -> note, this error now occurs in a different file
8) Delete "google/protobuf/" in the lines 66-75 of the opened file
9) Try to run the app again in XCode -> build succeeds and runs
10) Now you can also run the app from Android Studio and also build XCFrameworks for using them in a Flutter host app. -> Note: you will have to repeat these steps each time you perform flutter clean.

I know this is only a workaround but maybe this is enough of a hint for the actual issue so that @pauldemarco could kindly provide us with a fix. We would really appreciate that!
Thanks in advance.

chaochaox1990, 2020-07-19:
@aknolldev amazing

chaochaox1990, 2020-07-19:
@aknolldev 

[✓] Flutter (Channel stable, v1.17.5, on Mac OS X 10.15.4 19E287, locale
    zh-Hans-CN)
 
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.3)
[✓] Xcode - develop for iOS and macOS (Xcode 11.6)
[✓] Android Studio (version 3.6)
[✓] Connected device (1 available)

I change the code as your mention, XCode run all right but when I build ios-framework. some issue happened, could u know why?

admindeMacBook-Pro-3:car_control_sdk limi$ flutter build ios-framework --output=/Users/limi/Desktop/ios_framework

The following build commands failed:
        PBXCp /Users/limi/Downloads/flutter/.pub-cache/hosted/pub.flutter-io.cn/flutter_blue-0.7.2/ios/gen/Flutterblue.pbobjc.h
        /Users/Downloads/flutter/.pub-cache/hosted/pub.flutter-io.cn/flutter_blue-0.7.2/ios/gen/Flutterblue.pbobjc.h
(1 failure)

soyangel, 2020-07-31:
In my case I get this error:
```
% flutter build ios-framework --output=/tmp/frameworks 

Downloading ios tools...                                           12,2s
Downloading ios-profile tools...                                    7,9s
Downloading ios-release tools...                                   31,8s
Running "flutter pub get" in my_ios_plugin...                       6,5s
 
Building frameworks for com.example.myIosPlugin in debug mode...
 ├─Populating Flutter.framework...                                  1,1s
 ├─Building App.framework...                                        9,5s
Running pod install...                                              5,1s
 ├─Building plugins...                                             27,4s
Unable to build plugin frameworks: ** BUILD FAILED **

The following build commands failed:
	PBXCp /Users/angel/flutter/soft/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.2/ios/gen/Flutterblue.pbobjc.h
	/soft/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.2/ios/gen/Flutterblue.pbobjc.h
(1 failure)
```
Note that that the destination path starts with /soft/flutter/  but it should be /Users/angel/flutter/soft/flutter/.

soyangel, 2020-07-31:
Found a way to make it work for me.

On ios/flutter_blue.podspec replace:
```
ss.dependency "Protobuf", '~> 3.11.4'
```
with:
```
ss.dependency "Protobuf", '~> 4.0.0-rc1'
```

LeoAiolia, 2021-08-25:
same issue

LeoAiolia, 2021-08-31:
I changed flutter_blue ios depency to:

ss.dependency "Protobuf", '~> 3.12.0'

and work well.

一开始我将依赖改成了 ss.dependency "Protobuf", '~> 4.0.0-rc1'