pauldemarco/flutter_blue #630: Flutter blue hangs when reconnecting to device (Android)

CLDiego, 2020-07-10:
Hi,

I'm developing an app that communicates to a Microchip BLE module (RN4871) which has a transparent UART service (basically the module transmits through BLE the data it receives using a physical a serial interface). I've been having trouble when trying to reconnect.

This is the flow of the app:
1. Scan for devices and let user connect to the RN4871
2. Wait for connection
3. Obtain services and find the UART one, then look for the appropiate characteristic
4. Subscribe to notifications of the UART characteristic
5. Receive/Send data
6. Disconnect with a button, and wait for disconnection.

Here is when the problem starts, if I try to go through the same steps again, the app gets stucks at step 3. It finds the characteristic but it get's stuck trying to subscribe.

Also, oddly enough when this happens I get a prompt from the 'google assistant headphones' trying to access folders.

At the moment the solution is to close the app, turn on/off my BT adapter, and relaunch the app.

Any suggestions? should I unsubscribe before disconnecting? can this be the issue?

ericbouchard, 2020-07-27:
Got the same issue, i notice that i get a Timeout exception when trying to enable the notifications on the characteristic after reconnecting. In any case i do not get any new data from the characteristic.

borisllona, 2022-01-19:
Any update here?