pauldemarco/flutter_blue #632: [CRASH] - Can't build iOS (debug + release)

Ahmadre, 2020-07-14:
I wanna use this plugin, but I can't use it for iOS:

```bash
event sent after app closed: {id: 0, progressId: null, message: Running "flutter pub get" in amptech-app...}
event sent after app closed: {id: 0, progressId: null, finished: true}
Launching lib/main.dart on iPad von App in debug mode...
Automatically signing iOS for device deployment using specified development team in Xcode project: J3W7Z5NLZD
Xcode build done.                                           36,7s
Failed to build iOS app
Error output from Xcode build:
↳
    ** BUILD FAILED **
Xcode's output:
↳
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Platform/DataStructures/Bag.swift:176:9: warning: 'Hashable.hashValue' is deprecated as a protocol requirement; conform type 'BagKey' to 'Hashable' by implementing 'hash(into:)' instead
        var hashValue: Int {
            ^
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxBluetoothKit/BluetoothManager.swift:161:65: warning: 'index(where:)' is deprecated: renamed to 'firstIndex(where:)'
                                if let index = strongSelf.scanQueue.index(where: { $0 == scanOperationBox.value! }) {
                                                                    ^
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxBluetoothKit/BluetoothManager.swift:161:65: note: use 'firstIndex(where:)' instead
                                if let index = strongSelf.scanQueue.index(where: { $0 == scanOperationBox.value! }) {
                                                                    ^~~~~
                                                                    firstIndex
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Platform/DataStructures/Bag.swift:176:9: warning: 'Hashable.hashValue' is deprecated as a protocol requirement; conform type 'BagKey' to 'Hashable' by implementing 'hash(into:)' instead
        var hashValue: Int {
            ^
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxBluetoothKit/BluetoothManager.swift:161:65: warning: 'index(where:)' is deprecated: renamed to 'firstIndex(where:)'
                                if let index = strongSelf.scanQueue.index(where: { $0 == scanOperationBox.value! }) {
                                                                    ^
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxBluetoothKit/BluetoothManager.swift:161:65: note: use 'firstIndex(where:)' instead
                                if let index = strongSelf.scanQueue.index(where: { $0 == scanOperationBox.value! }) {
                                                                    ^~~~~
                                                                    firstIndex
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:27:5: warning: 'public' modifier is redundant for type alias declared in a public extension
        public typealias CompletableObserver = (CompletableEvent) -> ()
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:37:5: warning: 'public' modifier is redundant for static method declared in a public extension
        public static func create(subscribe: @escaping (@escaping CompletableObserver) -> Disposable) -> PrimitiveSequence<TraitType, ElementType> {
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:57:5: warning: 'public' modifier is redundant for instance method declared in a public extension
        public func subscribe(_ observer: @escaping (CompletableEvent) -> ()) -> Disposable {
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:81:5: warning: 'public' modifier is redundant for instance method declared in a public extension
        public func subscribe(onCompleted: (() -> Void)? = nil, onError: ((Swift.Error) -> Void)? = nil) -> Disposable {
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:111:5: warning: 'public' modifier is redundant for static method declared in a public extension
        public static func error(_ error: Swift.Error) -> Completable {
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:122:5: warning: 'public' modifier is redundant for static method declared in a public extension
        public static func never() -> Completable {
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:133:5: warning: 'public' modifier is redundant for static method declared in a public extension
        public static func empty() -> Completable {
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:153:5: warning: 'public' modifier is redundant for instance method declared in a public extension
        public func `do`(onError: ((Swift.Error) throws -> Void)? = nil,
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:178:5: warning: 'public' modifier is redundant for instance method declared in a public extension
        public func concat(_ second: Completable) -> Completable {
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:189:5: warning: 'public' modifier is redundant for static method declared in a public extension
        public static func concat<S: Sequence>(_ sequence: S) -> Completable
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:202:5: warning: 'public' modifier is redundant for static method declared in a public extension
        public static func concat<C: Collection>(_ collection: C) -> Completable
        ^~~~~~~
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Pods/MultiplatformBleAdapter/iOS/RxSwift/Traits/Completable.swift:215:5: warning: 'public' modifier is redundant for static method declared in a public extension
        public static func concat(_ sources: Completable ...) -> Completable {
        ^~~~~~~
[... 290 lines cut ...]
    4 warnings generated.
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Runner/AppDelegate.swift:11:9: warning: initialization of variable 'flutter_native_splash' was never used; consider replacing with assignment to '_' or removing it
        var flutter_native_splash = 1
        ~~~~^~~~~~~~~~~~~~~~~~~~~
        _
    While building module 'flutter_blue' imported from /Users/rebar/Documents/AMPTech/amptech-app/ios/Runner/GeneratedPluginRegistrant.m:22:
    In file included from <module-includes>:1:
    /Users/rebar/Documents/AMPTech/amptech-app/build/ios/Debug-iphoneos/flutter_blue/flutter_blue.framework/Headers/flutter_blue-umbrella.h:13:9: fatal error: 'FlutterBluePlugin.h' file not found
    #import "FlutterBluePlugin.h"
            ^~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    /Users/rebar/Documents/AMPTech/amptech-app/ios/Runner/GeneratedPluginRegistrant.m:22:9: fatal error: could not build module 'flutter_blue'
    @import flutter_blue;
     ~~~~~~~^~~~~~~~~~~~
    2 errors generated.
    note: Using new build system
    note: Building targets in parallel
    note: Planning build
    note: Constructing build description
Could not build the precompiled application for the device.

Error launching application on iPad von App.
Exited (sigterm)

```

chaochaox1990, 2020-07-15:
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    While building module 'flutter_blue' imported from /Users/limi/Desktop/aar_test/flutter_module_aar/.ios/Flutter/FlutterPluginRegistrant/Classes/GeneratedPluginRegistrant.m:14:
    In file included from <module-includes>:1:
    In file included from /Users/limi/Desktop/aar_test/flutter_module_aar/.ios/Pods/Target Support Files/flutter_blue/flutter_blue-umbrella.h:14:
    /Users/limi/Downloads/flutter/.pub-cache/hosted/pub.flutter-io.cn/flutter_blue-0.7.2/ios/gen/Flutterblue.pbobjc.h:13:10: fatal error: could not build module 'protobuf'
     #import "GPBProtocolBuffers.h"
      ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
    2 errors generated.
    /Users/limi/Desktop/aar_test/flutter_module_aar/.ios/Flutter/FlutterPluginRegistrant/Classes/GeneratedPluginRegistrant.m:14:9: fatal error: could not build module 'flutter_blue'
    #import <flutter_blue/FlutterBluePlugin.h>
     ~~~~~~~^
    3 errors generated.
    note: Using new build system
    note: Planning build
    note: Constructing build description

the same to me

joyfishzeyu, 2020-07-16:
the same to me

Ahmadre, 2020-07-16:
Guys if you got the same problem, please thumbs up the issue to get the maintainers aware of this. Thanks!

joyfishzeyu, 2020-07-16:
My problem has been solved (for the time being) [click](https://github.com/pauldemarco/flutter_blue/issues/514)

[](

[image]

Ahmadre, 2020-10-26:
This solves the compilation but the app is crashing after splashscreen...

dipcikibrahim, 2021-09-29:
I've got the same error.