pauldemarco/flutter_blue #634: How do I watch the DEVICE's state?

Zcating, 2020-07-17:
I have some problem with `BluetoothDevice.state`. It seems like a pull way not a push way. I want to watch the device's state, which api can I use for it.

Mirageofmage, 2020-07-18:
`BluetoothDevice.state` returns a `Stream<BluetoothDeviceState>`, By using `BluetoothDevice.state.listen` I can save the current `BluetoothDeviceState` as a variable. I have logic within a stateful widget that updates based on the variable.

Zcating, 2020-07-19:
@Mirageofmage  Thank you, but I already found this way to watch the state.  When you close the Bluetooth by Setting, the state stream would not notify you

Zcating, 2020-07-19:
`FlutterBlue.instance.state` , this stream will notify when Bluetooth closed.