pauldemarco/flutter_blue #637: Connected device disconnects on new scan request (iOS)

eriknorth, 2020-07-21:
Hi,

I am having an issue that if I have connected to a device and I start a new scan, the connection with the connected device is broken. I even do not get "disconnected" device state change. Seems like the object that holds the connection is destroyed when a new scan is started.

Is this possible to fix? Or it is not possible to scan for new devices while maintaining current connection.

eriknorth, 2020-07-21:
After debugging in XCode I noticed an error message:
`
[CoreBluetooth] API MISUSE: Forcing disconnection of unused peripheral <CBPeripheral: 0x280704960, identifier = 9F2C4F65-E692-88D4-1FD2-534E0F3CA3A6, name = device, state = connected>. Did you forget to cancel the connection?`

After checking the code, it seems that when performing a new scan, the reference to the previously connected device is removed by calling `[self.scannedPeripherals removeAllObjects];`. It looks like there is no implementation for storing strong reference to connected/used devices. 

I understand that this is probably a limitation in the library, however it should not cause "API MISUSE" and connected devices should be disconnected properly, when starting a new scan.

robinjohn, 2020-09-09:
Hi @eriknorth , may I know how you were able to solve the above error ? 
I got the same Xcode error message when trying to connect (on iOS v13.0 using flutter_blue's 0.7.2).

HninNweAye, 2022-03-03:
if (Platform.isIOS) {
        await Future<void>.delayed(const Duration(seconds: 1));
      }
      await bleReaderDevice?.connect(autoConnect: false);
      await _sendData();

I also had the exact same issue.Once I found my device in scanned result list then I added a delay 1 second before connecting to the device.