pauldemarco/flutter_blue #638: BLE Pair feature

shruti-techindustan, 2020-07-22:
can you please add a pair ble devices feature in the library? Actually, I want to pair the BLE devices in-app not from the Bluetooth setting.

ekuleshov, 2020-07-22:
What kind of device you wanted to pair? Usually you don't pair BLE devices, just connect.

manoelsrs, 2020-07-23:
@shruti-techindustan when you ask device to pair and try to connect, the box to input pin will appear.

ekuleshov, 2020-07-23:
With most ble devices you don't need to pair. Scan and then connect. The Bluetooth ones are different story. They aren't supported by flutter_blue

manoelsrs, 2020-07-23:
I agree with you, but I'm using a BLE device that needs to pair: https://shop.accu-chek.com/eShop/accu-chek-guide-meter-M005_us.html?cgid=Meters-Shop#start=1

eriknorth, 2020-07-23:
Pairing process is usually handled by the OS of the phone. During the pairing/bonding, OS will show a prompt with a pin input.
Device can request the phone to initiate pairing, but if it does not, then as soon as you will try to access a Characteristic that requires encrypted link or MITM protection, then the phone will request pairing/bonding and you should get the system level prompt.

rankush, 2020-08-05:
In My use case also BLE pairing is required. Like to provide the access of ANCS (Apple notification center service) of my iOS device (where I am running flutter app with flutter_blue lib) to other device ( say a smart watch).

rankush, 2020-08-05:
> Pairing process is usually handled by the OS of the phone. During the pairing/bonding, OS will show a prompt with a pin input.
> Device can request the phone to initiate pairing, but if it does not, [...]

Can we wrap the bond request and bond consent of OS to this flutter lib, so that flutter app will use that wrapper to listen OS specific callbacks for bonding/pairing

gumbypp, 2020-08-05:
> Can we wrap the bond request and bond consent of OS to this flutter lib, so that flutter app will use that wrapper to listen OS specific callbacks for bonding/pairing

As far as I know, this is not possible with iOS. Not sure about Android.

sudheer-palaparambil, 2020-09-11:
@manoelsrs did you have any success in reading data from Accu-Chek device ?

devaski, 2021-06-09:
I have tried to read accu-check guide. Actually, Flutter Blue, Android, the device requires us to choose the wireless > pair device option in the glucose device.. only then,  the device name iis listed in the available ble devices, with connect button.

I could read Charactertics as well With read button.. While I read, I am able to get the device name, manufacturer name , version name etc. I.e. this is achieved after printing String.fromCharcodes.

However, glucose value reads, Numbers are displayed but, when I convert,  I see an encrypted data?.
I am not sure, if that is the Glucose level data,

Any thoughts and guidance. 

Thanks in advance.
Deva

Secondly, I am not able to get the reference information for splitting these number to fields or to read the required data.

ekuleshov, 2021-06-10:
Not sure where to look for the specs on glucose meters. You can pick at the Nordic code at:

https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble-common/src/main/java/no/nordicsemi/android/ble/common/data/cgm/CGMSpecificOpsControlPointData.java

sudheer-palaparambil, 2021-06-13:
This is how I read the value. Source attached.
[SensorPage.dart.txt](https://github.com/pauldemarco/flutter_blue/files/6643580/SensorPage.dart.txt)

devaski, 2021-06-14:
Thank yo Mr.Sudheer

ivan2angelov, 2021-09-27:
Lack of BLE pairing is a real show stopper for **secure** BLE communication. In addition many applications would benefit from bonding.
Without pairing and bonding the use of flutter_blue is very limited and I would even claim the attractiveness of flutter for my company becomes questionable.

kiranruth, 2022-03-23:
Were you able to figure it out ? @devaski @sudheer-palaparambil ?

I tried with @sudheer-palaparambil code, but, that doesn't seem to work for me.

cah-aamir, 2022-09-08:
In Android one can subscribe to bond state change (BluetoothDevice.ACTION_BOND_STATE_CHANGED) and once you enter the PIN or accept the pairing request the state changes to BluetoothDevice.BOND_BONDED. This is how we determine a BLE device is paired before starting to write/access secure characteristics. I see none of the flutter BLE plugins currently support listening to bond state change. Unless I haven't looked enough?

devaski, 2022-10-11:
Dear Kiran

flutter_blue code work, however the BLE device Glucose test device i used
seems to be having their encryption. So beyond identifying the device, data
could not be read.

https://github.com/pauldemarco/flutter_blue/issues/638#issuecomment-858131233

Regards
Devaski

padma-araksha, 2023-05-05:
I am using flutter blue plus pluggin After the selection of bluetooth name when we have to pair bluetooth we have to enter the passkey.. What is the process of this passkey and how it generates the passkey and how it paired is there any Api to call

AymenXD, 2023-07-17:
@padma-araksha  any solution with the Flutter Blue plus Plugin. I encounter the same problem without knowing how to proceed!

padma-araksha, 2023-07-18:
List<int> value = await d.read();

 With this read function directly it calls the passkey function

ionutmurgu, 2023-08-03:
> List<int> value = await d.read(); With this read function directly it calls the passkey function
> […](#)

What do you mean by d.read()? Speciffically what is "d"?

Gaurav-CareMonitor, 2024-02-09:
Hi Team, Any luck on connecting with Accu Chek Guide?
Anyone able to get the data from device?

Gaurav-CareMonitor, 2024-02-14:
> **[devaski](/devaski) ** c

@devaski  Can you please share the code how you are able to get the encrypted data, it doesnt seem to be working for me.

devaski, 2024-02-14:
It has been some time
 Let me check the backups and update you. Thanks DG

Gaurav-CareMonitor, 2024-02-14:
Thanks @devaski , Would be a great help, Were you able to decode the readings as well?

devaski, 2024-02-14:
The limitation we have connecting OEM devices relates to kind of encryption
issues, or custom specific protocols used by the OEM.

Gaurav-CareMonitor, 2024-02-14:
Seems like it, can you still share the code that you did, would love to check it out and try to decrypt the readings if possible

iamongit, 2024-02-14:
> Thanks @devaski , Would be a great help, Were you able to decode the readings as well?

You need to get the official specs from the manufacturer, generally it's proprietary info and not available on the internet for everyone. The specs contain all the info you'll need to communicate with the device as well as decode the readings. You can try to contact the manufacturer or your supplier, describe your use case and request for the specs. Easier if your company deals with them and has ordered the devices in question.