pauldemarco/flutter_blue #644: Could not scan and connect the available devices on many devices

prakharGargIbirds, 2020-08-04:
The example given by the flutter_blue team is not running on many devices but running on some devices. The permissions are given fully but still, the app is not able to discover available devices, and if it can't scan, how it will show the available devices on the list, also I'm not able to discover the currently connected BLE device to my mobile Bluetooth and all these issues I'm facing in the android.
[image]
[image]

calvinsug, 2020-09-17:
I also face the exactly same situation for this, have you solved this issue ?

prakharGargIbirds, 2020-09-17:
> I also face the exactly same situation for this, after start scanning, the result is always 0 device founds.
> Have you solved this issue ?

The issue was resolved automatically after updating the package. Please try to update the package.

closed 2020-09-17 by prakharGargIbirds

calvinsug, 2020-09-17:
i think i have already used the latest version (0.7.2), but still facing this issue