pauldemarco/flutter_blue #646: Can't create BluetoothDevice object

JUREXT, 2020-08-05:
BluetoothDevice myDevice = BluetoothDevice(
   id: DeviceIdentifier("MAC address"),
   type: BluetoothDeviceType.le,
   name: "Device Name",
),
await myDevice.connect();

Hi,
I am using ^0.7.2 version and when I run on apple device my app, the logs don't show me the mac address, but I know the device mac address to connect to, the code above doesn't work for me. How can in create device in connect to it?
Problem: BluetoothDevice doesn't have a default constructor.
Thank you.

Also some info here: https://github.com/pauldemarco/flutter_blue/issues/333?fbclid=IwAR0CZEmN6-IjolSDBlSCoQmRzeNIEblfjpElk3MCwc80e54SkLS1BRGxsso

Fleximex, 2020-08-06:
https://github.com/pauldemarco/flutter_blue/issues/344

JUREXT, 2020-08-06:
Problem solved for now, need to test it on iphone.

1. Create a Enum

enum DeviceBtType {
  LE,
  CLASSIC,
  DUAL,
  UNKNOWN
}

2. Created method

BluetoothDevice createBluetoothDevice(String deviceName, String deviceIdentifier, DeviceBtType deviceType) {
    proto.BluetoothDevice p = proto.BluetoothDevice.create();
    p.name = deviceName;
    p.remoteId = deviceIdentifier;

    if(deviceType == DeviceBtType.LE) {
      p.type = proto.BluetoothDevice_Type.LE;
    } else if(deviceType == DeviceBtType.CLASSIC) {
      p.type = proto.BluetoothDevice_Type.CLASSIC;
    } else if(deviceType == DeviceBtType.DUAL) {
      p.type = proto.BluetoothDevice_Type.DUAL;
    } else {
      p.type = proto.BluetoothDevice_Type.UNKNOWN;
    }
    return BluetoothDevice.fromProto(p);
  }

3. Imports

import 'package:flutter_blue/flutter_blue.dart';
import 'package:flutter_blue/gen/flutterblue.pb.dart' as proto;
import 'device_type.dart';

JUREXT, 2020-08-06:
> #344

Thank you, it works.

RedsAnDev, 2020-08-06:
Thank you so much, I will test it for my application.

yuriboeira11tx, 2021-04-20:
@pauldemarco this is it?

AlexHartford, 2021-06-17:
@JUREXT Thank you!! It really would be great to have this built into the library, as in, a simpler way to create BluetoothDevices. Very busy lately but might make a PR one day.