pauldemarco/flutter_blue #647: Turn On / Turn Off Bluetooth

javiervalero31, 2020-08-07:
I only need know if is possible to turning on and turning off the Bluetooth from the flutter_blue lib.
If it is possible I need some examples to do this issue.

I have searched during several hours but I did not finded anything ! 
Can someone  help me, please?

JUREXT, 2020-08-07:
Find out the state:
- FlutterBlue.instance.isAvailable
- FlutterBlue.instance.isOn
Then use a package like: https://pub.dev/packages/system_setting and open settings.
Or a popup dialog to inform user to do something with the bluetooth.

javiervalero31, 2020-08-07:
Thanks.
I hope got it feature in the lib soon. Is the most important and will be friendly if the user could turning on or turning off the Bluetooth from the app and not from the Android Settings Screen.

@JUREXT  Does you know if the options you gave me are supported by iOS ?

javiervalero31, 2020-08-08:
> Thanks.
> I hope got it feature in the lib soon. Is the most important and will be friendly if the user could turning on or turning off the Bluetooth from the app and not from the Android Settings Screen.
> 
> @JUREXT Does you know if the options you gave me are supported by iOS ?

If it helps someone, i will respond to myself...  Yes, the system_settings is supported to iOS

closed 2020-08-08 by javiervalero31