pauldemarco/flutter_blue #648: Missing packets in notifications

itamardo, 2020-08-09:
I'm trying to work with a device that uses a BGX BLE SOC.
It sends a message of length 104 bytes which is received on a specific characteristic notifications on which I am listening.
I am using an MTU of 247 bytes.
It seems that the received message is divided into two packets, the first of about 60 (between 59 and 63) bytes and the second packet with what's left.
This works perfectly when using a native Android app and ble implementation. I receive two onCharacteristicChanged and both packets and am able to reconstruct the full 104-byte message.
However, with flutter (also on Android for now) I seem to only be getting the first 60-byte notification and never the second one.
I ran with LogLevel.debug and saw that indeed only one onCharacteristicChanged is logged instead of the expected 2.
Is there some configuration I need to use or some other mechanism except the setNotifyValue/listen mechanism?
I am using version 0.7.2 and flutter stable 1.20

ayushin, 2020-08-18:
Similar problem on IOS. Any news on this?

ajaymore, 2020-08-24:
I have encountered the same issue on Android.

ajaymore, 2020-08-24:
I could solve the issue by increasing the mtu size. Please check if mtu is the problem.

```
device.requestMtu(248);
```

itamardo, 2020-08-24:
No. I used an MTU of 247 (on Android only because iOS doesn't allow IMU setting. iOS uses an MTU of about 185). That's not it