pauldemarco/flutter_blue #655: Listeners triggered on two different services with same Characteristic UUID

fdugdale123, 2020-08-13:
I am using a custom service to control a light. the service contains one characteristic for the light state, I have multiple lights and therefore am using multiple instances of services with the same service UUID and characteristic UUID. When I send data from the peripheral device through the state characteristic of one service it activates the listener on both services.

RedsAnDev, 2020-08-16:
In these months I worked a lot on a scenario like yours. I choose to save the characteristic in a Map, for each device and I'm able to get different data from each device, without an effect like yours. I'm using notify-feature and listen to each stream on at least 4 different devices.

fdugdale123, 2020-08-17:
> In these months I worked a lot on a scenario like yours. I choose to save the characteristic in a Map, for each device and I'm able to get different data from each device, without an effect like yours. I'm using notify-feature and listen to each stream on at least 4 different devices.

would you be ab

fdugdale123, 2020-08-17:
> In these months I worked a lot on a scenario like yours. I choose to save the characteristic in a Map, for each device and I'm able to get different data from each device, without an effect like yours. I'm using notify-feature and listen to each stream on at least 4 different devices.

I have not tried connecting on multiple devices so I do not know if this issue persists if I were to connect to multiple. my issue is that I have multiple services with a characteristic with the same UUID on the same device.

fdugdale123, 2020-08-17:
just tried your solution to save in a map and am still getting the same result. @pauldemarco any idea what causes this issue?

RedsAnDev, 2020-08-17:
> I have not tried connecting on multiple devices so I do not know if this issue persists if I were to connect to multiple. my issue is that I have multiple services with a characteristic with the same UUID on the same device.

Whoa, that's sound new for me. I google a few query and I found that is a common policy for many sensors. At this link
https://stackoverflow.com/questions/27413272/what-gatt-services-use-multiple-characteristics-with-the-same-uuid
you can find a docs about BLE standard.

fdugdale123, 2020-08-17:
again, this is not my issue. the issue I am having is that I have multiple of the same service with the same characteristic. not one service with multiple characteristics with the same UUID. 

my case:
- Light Service
        Characteristic
- Light Service
        Characteristic

the link:
- Light Service
        Characteristic
        Characteristic

in my case when a value is changed on one characteristic the listener in both services is activated

mgeilich, 2023-02-28:
I have this same problem.  I have a device with two characteristics with the same UUID.  Writing to either characteristic only writes to the first service.

Note that I have native code for both Android and iOS that works perfectly for this, so the device is not the problem.  In my native code, writing to each characteristic writes to a separate service.