pauldemarco/flutter_blue #656: Project failed to compile flutter blue with Flutter 1.20

zuboje, 2020-08-14:
IF flutter is upgraded to Flutter 1.20.x version project will fail to compile saying flutter blue uses deprecated API:

....AppData\Local\Pub\Cache\hosted\pub.dartlang.org\flutter_blue-0.7.2\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: ....build\flutter_blue\generated\source\proto\debug\javalite\com\pauldemarco\flutter_blue\Protos.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.

akashpamal, 2020-08-14:
I had the same problem. My setup was:
Flutter version 1.20.2
Dart version 2.9.1

RedsAnDev, 2020-08-16:
Good evening, I have the same package version, but different setup and **I don't get these warning's messages**.

**Flutter_blue version from pubspec.lock**

> flutter_blue:
>     dependency: "direct main"
>     description:
>       name: flutter_blue
>       url: "https://pub.dartlang.org"
>     source: hosted
>     version: "0.7.2"

**dart --version output**

> Dart SDK version: 2.9.0 (stable) (Fri Jul 31 10:59:48 2020 +0200) on "macos_x64"

**flutter doctor output**
> 
> [✓] Flutter (Channel beta, 1.20.2, on Mac OS X 10.15.6 19G73, locale it-IT)
>  
> [✓] Android toolchain - develop for Android devices (Android SDK version 30.0.1)
> [✓] Xcode - develop for iOS and macOS (Xcode 11.6)
> [✓] Android Studio (version 4.0)
> [✓] IntelliJ IDEA Ultimate Edition (version 2020.1.2)
> 
>

akashpamal, 2020-08-18:
It looks like the problem occurs when doing a FULL RESTART on Android Studio AFTER already working in Android Studio for a while. I was able to fix the problem by doing the following steps:
1. Avoid a full restart unless necessary; try to use hot reload and hot restart (this doesn't seem to trigger the problem).
2. If the problem occurs, cd to your project folder in terminal and type "**flutter clean**". This deletes the auto-generated files for iOS builds, and cleaning them will ensure that they are rebuilt fully and correctly.
3. In Android Studio, click **File -> Invalidate Caches / Restart**. This just clears the cache in Android Studio and restarts the application.
Let me know if that helps!

zuboje, 2020-08-18:
@akashpamal, in the long run, this is a workaround but not a solution, deprecated APIs should be updated

akashpamal, 2020-08-18:
Yup, that's true! I was just offering a temporary fix so we can continue developing while the APIs are updated.

safield, 2020-10-21:
Same issue here with Flutter 1.22.2 and Dart 2.10.2.